xqms/rosmon #9: rosmon in ROS Indigo

lucascoelhof, 2018-04-10:
Hello,

First of all, thanks for making this great project available!

I need to compile rosmon for a target that only runs ROS Indigo. Since rosmon is not available on the repositories for ROS Indigo, I tried to compile it by myself but I faced some problems:

1. **libcurses path**: CMake was unable to find the path of Curses, and I had to modify the CMakeLists.txt to allow it to find. I've added these lines to the top of the file. 
```
set(CURSES_LIBRARY "/opt/lib/libncurses.so")
set(CURSES_INCLUDE_PATH "/opt/include")
```
After adding these lines, CMake could find Curses correctly. Is there a better solution for this?

2. **QComboBox**: CMake could not find "setCurrentText" 

```
/home/lucas/catkin_ws/build/rosmon/ui_mon_gui.h: In member function ‘void Ui_MonGUI::retranslateUi(QWidget*)’:
/home/lucas/catkin_ws/build/rosmon/ui_mon_gui.h:79:18: error: ‘class QComboBox’ has no member named ‘setCurrentText’
         nodeBox->setCurrentText(QApplication::translate("MonGUI", "[auto]", 0, QApplication::UnicodeUTF8));
```
Very weird, I've cheched and I have qt5dev installed. I've also compiled other qt5 applications on my computer with no problem. 

3. **ros::SteadyTimer**: I could also not find SteadyTimer class.

```
In file included from /home/lucas/catkin_ws/src/rosmon/src/monitor/monitor.cpp:4:0:
/home/lucas/catkin_ws/src/rosmon/src/monitor/monitor.h:70:2: error: ‘SteadyTimer’ in namespace ‘ros’ does not name a type
  ros::SteadyTimer m_statTimer;
```

From the reserach I've made, the SteadyTimer class was introduced in ROS Kinetic (not exactly sure about that, but you can compare these two pages http://docs.ros.org/indigo/api/roscpp/html/annotated.html http://docs.ros.org/kinetic/api/roscpp/html/annotated.html)

Considering the problems faced, do you think it would be possible to have rosmon in ROS Indigo too?

xqms, 2018-04-10:
Thanks for the interest in rosmon!

I made the choice to drop Indigo support some time ago, since Kinetic is the current LTS release. You may have luck compiling a slightly older version, maybe revision 9dab795c6dd5a5d2b3b24b574905665fdd133880 will work out of the box.

Regarding your specific issues:

1) *ncurses*: Under Ubuntu, installing `libncurses5-dev` should be enough. There is a small issue that requires deletion of the CMake cache (just delete the build directory) and building again to detect the new ncurses installation.
Otherwise, you can specify custom ncurses paths on the command line:
`cmake -DCURSES_LIBRARY=/opt/lib/libncurses.so -DCURSES_INCLUDE_PATH="/opt/include"`
should work with your settings.

2) *QComboBox*: This is weird. Can you tell me your Qt version? In Qt 5.6, 5.9 and 5.10 this method is present. Maybe rosmon is linking against Qt4? Then, I definitely recommend trying an older version.

3) `ros::SteadyTimer` was introduced with ROS Kinetic, but you can replace it with `ros::WallTimer` without problems (the CPU load measurement will be inaccurate while changing the date/time on the computer).

Again, older versions should compile just fine under Indigo.

lucascoelhof, 2018-04-12:
1. ncurses: Yes, you are right, deleting build directory after installing libncurses5-dev solved the problem!
2. QComboBox: After deleting the build directory it also solved my problem. Even though I had Qt already on my computer and I was able to build other Qt binaries, I had to install some more libraries to build rosmon. Maybe that was the problem...
3. Thanks for adding support for Indigo! It worked nicely :+1:

closed 2018-04-12 by lucascoelhof