xqms/rosmon #16: Errors loading move_group planning adapters

BrettHemes, 2018-04-30:
For my use case I have no issues when launching with `roslaunch` but `rosmon` results in errors when initializing the `move_group` planning pipeline. I get the below for each default adapter in `default_planner_request_adapters` (four in all), resulting in my joint_trajectory_action server never coming up.

      move_group: [PlanningPipeline::configure]: Exception while loading planning adapter plugin 'default_planner_request_adapters/AddTimeParameterization
      move_group: 				': According to the loaded plugin descriptions the class default_planner_request_adapters/AddTimeParameterization
      move_group: 				 with base class type planning_request_adapter::PlanningRequestAdapter does not exist. Declared types are  default_planner_request_adapters/AddTimeParameterization default_planner_request_adapters/Empty default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints default_planner_request_adapters/FixWorkspaceBounds industrial_trajectory_filters/AddSmoothingFilter industrial_trajectory_filters/NPointFilter industrial_trajectory_filters/UniformSampleFilter

FYI I am pushing EVERYTHING down into a separate namespace via `<group ns=...>` tag in the high-level launch files in case that makes a difference.

xqms, 2018-04-30:
This may be an issue with white space stripping from parameters - roslaunch does a whole lot of magic there. The log message seems to indicate that there is a `\n` character at the end of the plugin type. Is there are chance that you can find the `default_planner_request_adapters/AddTimeParameterization` string in your launch files?

I'm not really familiar with the `move_group` pipeline.

xqms, 2018-04-30:
Stab in the dark, but #17 may help. If you have time, could you test it?

xqms, 2018-05-01:
Merged #17.

closed 2018-05-01 by xqms