xqms/rosmon #19: Segmentation fault

MichaelT64, 2018-05-02:
I tried to start the following launch file:
<launch>
  <node pkg="rosserial_python" type="serial_node.py" name="robby_node">
    <param name="port" value="/dev/serial0" />
    <param name="baud" value="500000" />
  </node>
  <node pkg="tf" type="static_transform_publisher" name="sonar_broadcaster" args="0.07 0 0.045 0 0 0 1 base_link sonar 100" />
</launch>

mon exited with segmentation fault. When removing the second node it works.

xqms, 2018-05-02:
Hmm, I can't reproduce this, neither with master nor with the released binary packages (1.0.2). Can you give me more information?

Which ROS version are you on? Did you build rosmon from source or are you using binary packages?

Could you run
```
gdb -ex run --args /opt/ros/kinetic/lib/rosmon/rosmon path/to/my/launch_file.launch
```
and once the crash occurs, obtain a backtrace with `bt`?

xqms, 2018-05-03:
Ah okay, that probably means we have a problem on ARM architecture. I'll try to set up an emulator to test things here.

MichaelT64, 2018-05-03:
Don't know if you found something in the meantime, but here is the output of the gdb:

```
Starting program: /opt/ros/kinetic/lib/rosmon/rosmon /home/husarion/ros_workspace/src/robby/robby.launch
Cannot parse expression `.L966 4@r4'.
warning: Probes-based dynamic linker interface failed.
Reverting to original interface.

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Loaded launch file in 0.129969s
[New Thread 0x7470a420 (LWP 7906)]
[New Thread 0x73f0a420 (LWP 7907)]
[Thread 0x7470a420 (LWP 7906) exited]
[New Thread 0x7370a420 (LWP 7908)]
[Thread 0x73f0a420 (LWP 7907) exited]
[New Thread 0x72f0a420 (LWP 7909)]
[Thread 0x7370a420 (LWP 7908) exited]
[Thread 0x72f0a420 (LWP 7909) exited]
ROS_MASTER_URI: 'http://localhost:11311'
roscore is already running.
[New Thread 0x72f0a420 (LWP 7912)]
[New Thread 0x7370a420 (LWP 7913)]
[New Thread 0x73f0a420 (LWP 7914)]
[New Thread 0x7470a420 (LWP 7919)]
Running as '/rosmon_1525374476233113050'
[New Thread 0x723ff420 (LWP 7920)]
[New Thread 0x71bff420 (LWP 7921)]

Thread 1 "rosmon" received signal SIGSEGV, Segmentation fault.
0x00057504 in rosmon::UI::drawStatusLine() ()
(gdb) bt
#0  0x00057504 in rosmon::UI::drawStatusLine() ()
#1  0x20202020 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 

```

xqms, 2018-05-03:
I can reproduce this using qemu. Thanks for the backtrace, I'll investigate this further...

xqms, 2018-05-04:
Fixed in #24. I'll create a new release - it will take some time (weeks) to show up in the binary repositories, though.

closed 2018-05-04 by xqms

MichaelT64, 2018-05-05:
Thanks, is there a way to build this by myself? I don't want to wait weeks until I can use the new version ;-)

xqms, 2018-05-05:
Of course! You can clone always rosmon into a catkin workspace and compile it.

I have a slightly more convenient option for you: I just released version 1.0.3 and the ros_buildfarm already built a debian package for you:
http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-kinetic-rosmon/
This one should be the correct one for a Raspberry (armhf):
http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-kinetic-rosmon/ros-kinetic-rosmon_1.0.3-0xenial-20180505-110800-0800_armhf.deb

Just download and install with `sudo dpkg -i FILE`.

It will just take a few weeks until it's in the official repositories.

MichaelT64, 2018-05-10:
Works like a charm!
Thanks for your help.

PR #24 (2018-05-04): ui: calculate node name padding correctly on 32 bit architectures
Sometimes, C is a pain.
Fixes #19.
commit 595874cd (2018-05-04): ui: calculate node name padding correctly on 32 bit architectures

Sometimes, C is a pain.
Fixes #19.