xqms/rosmon #20: rosmon exits if parameter in <node> block contains tilde

MatthiasNieuwenhuisen, 2018-05-03:
Blocks of the structure
```
<node name="NODENAME" ...>
  <param name="~PARAMNAME" value="VALUE"/>
</node>
```
work in roslaunch but not in rosmon, while
```
<node name="NODENAME" ...>
  <param name="PARAMNAME" value="VALUE"/>
</node>
```
works as expected.

According to the ROS naming conventions, the tilde is redundant in this case, as the parameter inside of a node block is already private, but not forbidden.

Whereas roslaunch creates a parameter /NODENAME/PARAMNAME in both cases, rosmon exits with
```
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  Character [~] at element [14] is not valid in Graph Resource Name [/NODENAME/~PARAMNAME].  Valid characters are a-z, A-Z, 0-9, / and _.
```

xqms, 2018-05-03:
Thanks - should be fixed in #21. I'll merge as soon as I test this in more complex systems.

xqms, 2018-05-03:
Merged #21. I'll create a new release as soon as #19 is fixed, but new Debian packages will come with the next package sync (~ three weeks).

closed 2018-05-03 by xqms