xqms/rosmon #26: Support for multiple machines

lorenardi, 2018-05-08:
At our lab, we would like to use your tool to monitor ROS on multiple machines.
However, it doesn't look to be possible at the moment as the ``<machine>`` tag is not supported yet.
Are you planning to keep developing the tool to make it work on multiple machines in the next future?
Otherwise, we may consider to implement this functionality ourselves.

If someone else is interested or planning to implement it, please comment here, so we can put some effort together and avoid to replicate the work.

Thanks!

xqms, 2018-05-08:
Support for multiple machines is currently not planned. I'd be happy to receive PRs, though.

rosmon is mostly used in our lab in multi-master systems connected using [nimbro_network][1] - over lossy / intermittent WiFi networks. In this situation, central launching is not very useful, that's why we did not implement it so far.

We usually `ssh` into our robots and start a rosmon instance inside `tmux`. After starting, the rosmon ROS interface is used to monitor things remotely.

Do you have a particular use case in mind? Do you start multiple robots at once? Or do you just distribute nodes over machines inside one robot? I'm just curious ;-)

[1]: https://github.com/AIS-Bonn/nimbro_network

lorenardi, 2018-05-14:
Typically, in our lab we distribute nodes over multiple machines, so that each machine runs a set of functionalities of the robot. For example, one machine takes care of reading from the sensors, another one processes the data, etc. This allows to distribute computational resources, but also to avoid that changing something on a central machine might affect some basic functionalities of the robot. Thus, we run the robot by `ssh`  into the master and running from it all of the nodes over the machines.

I like rosmon :) so I'd like to put some effort to make it work over multiple machines.
Hope to have some time soon to implement it!

xqms, 2018-05-14:
I see. In that case, it might be as easy as putting `ssh user@machine` in front of the executed command in `src/monitor/node_monitor.cpp`. But some nice characteristic rosmon features will not work, e.g. automatic core dump collection. That would require some support (rosmon runner) on the remote side as well.

amilcarlucas, 2020-05-19:
or you could try node_manager_fkie