xqms/rosmon #28: Global remappings not working

MatthiasNieuwenhuisen, 2018-05-11:
Remappings in the launch or group scope are not applied to node topics, the original topics stay unremapped.

Launch file:
```
<launch rosmon-name="rosmon_uut">
        <!-- Test global remapping -->
        <remap from="original_test_input" to="remapped_test_input" />
        <remap from="original_test_output" to="remapped_test_output" />
        <node name="test2" pkg="rosmon" type="test_node.py">
                <remap from="~input" to="/original_test_input" />
                <remap from="~output" to="/original_test_output" />
        </node>
</launch>
```

The following test succeeds with roslaunch and fails with rosmon:
```
       def test_global_remapping(self):
                pub = rospy.Publisher('/remapped_test_input', String, queue_size=5)

                wfm = _WFM()
                sub = rospy.Subscriber('/remapped_test_output', String, wfm.cb)

                time.sleep(1);

                self.assertGreater(pub.get_num_connections(), 0)
                self.assertGreater(sub.get_num_connections(), 0)
                pub.publish('Hello world!')

                timeout_t = time.time() + 5
                while wfm.msg is None:
                        rospy.rostime.wallsleep(0.01)
                        if time.time() >= timeout_t:
                                self.fail('No reply to test message')

                self.assertEqual(wfm.msg.data, 'Hello world!')

                sub.unregister()
                pub.unregister()
```

Expected result:
```
# rostopic list
/remapped_test_input
/remapped_test_output
```

Actual result:
```
# rostopic list
/original_test_input
/original_test_output
```

xqms, 2018-05-13:
#30 should fix your issues (at least it passes your integration test and my unit tests). Could you test with your "launch files from hell"? ;-)

MatthiasNieuwenhuisen, 2018-05-14:
The patch works for me. I'll let you know when I find another "launch file from hell" ;-)

commit 2b41a34b (2018-05-14): launch: collect remappings in ParseContext

Essentially, <remap> behaves like <env>, so treat it the same way.
This fixes #28.
commit 854cc83f (2018-05-14): Merge pull request #30 from xqms/feature/global_remap

Support <remap> in other scopes than <node>. Fixes #28.
closed 2018-05-14 by xqms