xqms/rosmon #33: YAML handling of quotes discrepancy

BrettHemes, 2018-06-06:
There seems to be a difference when loading YAML files in which quoted values are not interpreted the same as with `roslaunch`.  This results in type issues when invoking `rosparam` (esp. when using `XmlRpc` objects).

Use case is using `<rosparam command="load" file="path/to/config.yaml"/>` in the launch file.

Quoted numbers (e.g., my_param: "76") were still loading as ints (instead of strings per the spec?).  

Ultimately this was found due to me accidentally quoting some numeric parameters and has now been resolved on my end by simply removing quotes and having better type checking / assertions when parsing `XmlRpc` objects but thought it worth reporting.

xqms, 2018-06-06:
Interesting, thanks for the report. How to convert YAML types to XmlRpc types is not fully specified, as far as I know - but we should definitely match roslaunch's behavior. I'll look into this in the next days.

The problem might be that yaml-cpp, which we use for YAML parsing, hides types of scalars from us - if they are not specified using YAML tags. Especially, it might be hard to determine whether there were quotes surrounding the value...

xqms, 2018-06-10:
Should be fixed in #34.

PR #34 (2018-06-10): YAML quoted strings
Make sure that we create string parameters for quoted YAML scalars. This fixes #33.
closed 2018-06-10 by xqms