xqms/rosmon #44: Not possible to start rosmon (without ui) within a running rosmon session?

romainreignier, 2018-09-18:
First, let me thank you for `rosmon`, it has been a pleasure to use it for several months.
The features to stop and restart a node is quite nice.

In our robot, we need to launch nodes on the fly so, until now, we start a `roslaunch` command with a `QProcess`. I have tried to replace `roslaunch` to `rosmon` with `--disable-ui` argument but I have some issues.
- It cannot launch a Python node because of `Permission denied`
- And some nodes cannot find some parameters written in `<rosparam>` entries in the launch file.

So before digging into these issues, is it something known that rosmon instances cannot be run inside a rosmon one?

xqms, 2018-09-18:
> First, let me thank you for rosmon, it has been a pleasure to use it for several months.
The features to stop and restart a node is quite nice.

Thanks :-)

Without knowing more, it's a bit hard to debug. `--disable-ui` is definitely the way to go, though.

> It cannot launch a Python node because of Permission denied

Not sure what is happening there. Can you upload a full log?

> And some nodes cannot find some parameters written in <rosparam> entries in the launch file.

That sounds interesting, we always try to find and fix these inconsistencies. Did you compile rosmon from the current GitHub master or are you using the binary packages? We just fixed a few of these problems, so it might be worth it to compile from source. It always takes some time for these fixes to arrive in the debian packages...

If it's also happening in the current master version, I'd appreciate an example launch file :-)

romainreignier, 2018-09-18:
Hi,

> Without knowing more, it's a bit hard to debug
Yes, my post was not to ask for debug, but only to see if it should work or not, to not spend time on a feature that is not supposed to work :)

> Did you compile rosmon from the current GitHub master or are you using the binary packages? 
I am using the binary package version. I will try with git master version so.

romainreignier, 2018-09-18:
Ok, so I have tried to use the git version and I had the same results.

**BUT**, digging a bit further, I have noticed that the `launch` files I was loading on the fly did not have a specified namespace. With roslaunch, all the nodes in the launch file inherit the namespace of the node starting them and not with rosmon. So adding a `<group ns="my_ns">` in the launch file fixed the parameters issues but make it not compatible with roslaunch.

About the Python issue, I had the following error:

    p_programs_launcher: Could not execute '/home/romain/ws/src/p_programs_launcher/src/controller __name:=p_programs_launcher ~running_program:=/head/running_program ': Permission denied

But it occurred that the package `p_programs_launcher` was organized as follow:

    ├── CMakeLists.txt
    ├── package.xml
    ├── scripts
    │   └── controller
    ├── setup.py
    └── src
        └── controller
            ├── __init__.py
            └── main.py

So the Python executable `scripts/controller` had the same name as a directory in `src`. I have renamed `scripts/controller` to `scripts/programs_launcher` and **everything worked fine**.

I am not sure about that issue, if it is an issue on my side with the confusing naming or an issue on `rosmon` side.

xqms, 2018-09-18:
Thanks for the detailed analysis, that was very helpful. I created separate issues #45 and #46 for the two problems. I'll close this one when the two sub-issues have been solved.

xqms, 2018-09-19:
#45 and #46 have been fixed, closing this now. If an issue remains, feel free to reopen/comment.

closed 2018-09-19 by xqms