xqms/rosmon #45: Ignore directories while searching for node executables

xqms, 2018-09-18:
Quoting from #44:

> ```
> p_programs_launcher: Could not execute '/home/romain/ws/src/p_programs_launcher/src/controller __name:=p_programs_launcher ~running_program:=/head/running_program ': Permission denied
> ```
> 
> But it occurred that the package `p_programs_launcher` was organized as follow:
> 
> ```
> ├── CMakeLists.txt
> ├── package.xml
> ├── scripts
> │   └── controller
> ├── setup.py
> └── src
>     └── controller
>         ├── __init__.py
>         └── main.py
> ```
> 
> So the Python executable `scripts/controller` had the same name as a directory in `src`. I have renamed `scripts/controller` to `scripts/programs_launcher` and **everything worked fine**.
> 
> I am not sure about that issue, if it is an issue on my side with the confusing naming or an issue on `rosmon` side.

Seems that we should search for executable *files* only and ignore directories.

romainreignier, 2018-09-19:
I confirm that that commit fixed my issue.

xqms, 2018-09-19:
Thanks for the confirmation! I have merged #47.

closed 2018-09-19 by xqms