xqms/rosmon #49: Leading slash in namespace (ns=)

romainreignier, 2018-09-19:
When running the [MIT Racecar project](https://github.com/mit-racecar/racecar-simulator) on simulation, it starts with `roslaunch` but not `rosmon`.

Here is the log with `rosmon`:

                gazebo: [ServiceClientLink::handleHeader]: received a tcpros connection for a nonexistent service [//racecar/controller_manager/load_controller].
    controller_manager: [main]: Controller Spawner: Waiting for service controller_manager/switch_controller
                gazebo: [ServiceClientLink::handleHeader]: received a tcpros connection for a nonexistent service [//racecar/controller_manager/switch_controller].
    controller_manager: [main]: Controller Spawner: Waiting for service controller_manager/unload_controller
                gazebo: [ServiceClientLink::handleHeader]: received a tcpros connection for a nonexistent service [//racecar/controller_manager/unload_controller].
    controller_manager: [main]: Loading controller: left_rear_wheel_velocity_controller
                gazebo: [ServiceClientLink::handleHeader]: received a tcpros connection for a nonexistent service [//racecar/controller_manager/load_controller].
    controller_manager: Traceback (most recent call last):
    controller_manager:   File "/opt/ros/melodic/lib/controller_manager/spawner", line 209, in <module>
    controller_manager:     if __name__ == '__main__': main()
    controller_manager:   File "/opt/ros/melodic/lib/controller_manager/spawner", line 187, in main
    controller_manager:     resp = load_controller(name)
    controller_manager:   File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 439, in __call__
    controller_manager:     return self.call(*args, **kwds)
    controller_manager:   File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 509, in call
    controller_manager:     raise ServiceException("unable to connect to service: %s"%e)
    controller_manager: rospy.service.ServiceException: unable to connect to service: remote error reported: received a tcpros connection for a nonexistent service [//racecar/controller_manager/load_controller].
    controller_manager: [shutdown]: Shutting down spawner. Stopping and unloading controllers...
    controller_manager: [shutdown]: Stopping all controllers...
                gazebo: [ServiceClientLink::handleHeader]: received a tcpros connection for a nonexistent service [//racecar/controller_manager/switch_controller].
    controller_manager: [shutdown]: Controller Spawner error while taking down controllers: unable to connect to service: remote error reported: received a tcpros connection for a nonexistent service [//racecar/controller_manager/switch_controller].

You can notice the double slash `//` in the service name.
This double slash is due to the `ns="/"` here:

https://github.com/mit-racecar/racecar-simulator/blob/effd29f9edc20a03c8a1f666fb6fb6c0cbbee8ff/racecar_gazebo/launch/racecar.launch#L19

And the leading slash in `ns="/racecar"` here:

https://github.com/mit-racecar/racecar-simulator/blob/effd29f9edc20a03c8a1f666fb6fb6c0cbbee8ff/racecar_control/launch/racecar_control.launch#L10

Removing both make the system work with `rosmon`.

xqms, 2018-09-19:
Thanks for the report!

The solution is probably to compress double slashes in `ParseContext::enterScope()`. I'll check.

xqms, 2018-09-19:
I'm not familiar with `mit-racecar`, but it starts successfully now that #53 is merged. Closing.

closed 2018-09-19 by xqms

romainreignier, 2018-09-19:
Indeed, it works now for me too and on another project I had the same issue.
Thanks