xqms/rosmon #56: Show ROS standard output

Julz-s, 2018-10-22:
Hi,

I'm currently starting to use rosmon as an alternative to roslaunch - and it works well. However, one thing I'm missing is the ROS standard output (INFO, WARN, ERROR). With roslaunch, they appear right in the console and they are often quite helpful.

Is it possible to show them in rosmon too?

Best,
Julz

xqms, 2018-10-22:
The output of the `ROS_INFO/WARN/ERROR` macros should be shown in rosmon. If not, you have discovered a bug. Could you give me a bit more information about your setup?
Best would be a launch file, and outputs of roslaunch and rosmon, respectively.

Julz-s, 2018-10-24:
So, 
I've checked all the outputs again - and there is no bug. Because I launch quite a few nodes, there is a lot of output and I just missed the INFO messages (they are not labeled with [INFO] so I just overlooked them) :)

However, I discovered that having respawn="true" in a launch file does not cause a node to restart when I kill it in rosmon. Is this behavior intended?

Best,
Julz

xqms, 2018-10-24:
> However, I discovered that having respawn="true" in a launch file does not cause a node to restart when I kill it in rosmon. Is this behavior intended?

By "kill it in rosmon" you mean stopping the node through the rosmon user interface? The `respawn` flag is only intended for automatic restart in case of crashes (unnatural exits with exit code != 0). Manual starting/stopping through the rosmon UI always overrides this logic, otherwise there would be no way to permanently stop a node.

I'm closing this ticket since the issue seems resolved. If not, feel free to reopen.

closed 2018-10-24 by xqms