xqms/rosmon #57: Launch file difficulties

Errogant, 2018-10-24:
Hello!

I'm loving the tool and would like to do system resource usage monitoring using it. However, I'm encountering difficulties with a .launch file that was fine when parsed by roslaunch. I've searched around and seen that there have been a number of issues with XML parsing that have been resolved, but I haven't seen mine addressed. Please let me know if I missed something!

I'm getting a failure on the first line of the excerpt that "<arg> inside include needs name and value", which I looked for in the source code and found [here](https://github.com/xqms/rosmon/blob/1989e2e142a4252761469a83315d696559068a93/src/launch/launch_config.cpp)

I'm thinking this might just be because I have default instead of value as a field? Is there a reason this can't be default? 

Thanks! 

The launch file excerpt : 
```xml
  <arg name="kinect" default="false" />
  <arg name="xtion" default="false" />
  <arg name="camera_link_pose" default="0.15 0.075 0.5 0.0 0.7854 0.0"/>
  <include file="$(find sawyer_moveit_config)/launch/move_group.launch">
    <arg name="kinect" value="$(arg kinect)" />
    <arg name="xtion" value="$(arg xtion)" />
    <arg name="camera_link_pose" value="$(arg camera_link_pose)"/>
    <arg name="allow_trajectory_execution" value="true"/>
    <arg name="fake_execution" value="false"/>
    <arg name="info" value="true"/>
    <arg name="debug" value="$(arg debug)"/>
  </include>
```

xqms, 2018-10-24:
Your excerpt runs fine with rosmon here (I don't have `sawyer_moveit_config`, so I did not test for failures inside that package).

Could you please include the *full* error message? It should also print file and line number information.

This type of issue usually happens when you use `<arg name="..." default="..." />` *inside* a `<include>` tag, which does not make any sense.
roslaunch allows this and parses it as `<arg name="..." value="..." />`. We should probably do the same, even if it's against the spec. We could issue a warning in that case...

xqms, 2018-10-24:
If you like, you could test #58 on your setup. It should issue a warning and continue like roslaunch.

xqms, 2018-10-24:
For the record, here a problematic line in `sawyer_moveit_config`, which may cause this issue:
https://github.com/RethinkRobotics/sawyer_moveit/blob/bd38e352ee32cabccc9aa0bc45d838f17941b541/sawyer_moveit_config/launch/move_group.launch#L41

Errogant, 2018-10-24:
This works perfectly! Thanks for being so quick to respond

closed 2018-10-24 by Errogant