xqms/rosmon #59: Handle SIGTERM signal

romainreignier, 2018-10-25:
`rosmon` already handles `SIGINT` signal here:

https://github.com/xqms/rosmon/blob/edf2742fa44234a67be0ea527be8dbe5138b41e2/src/main.cpp#L389

But `roslaunch` also handles `SIGTERM` and `SIGUP` signals as seen here:

https://github.com/ros/ros_comm/blob/89ca9a7bf288b09d033a1ab6966ef8cfe39e1002/tools/roslaunch/src/roslaunch/pmon.py#L162

One use case I have is when a `rosmon` instance is launched within a [QProcess](http://doc.qt.io/qt-5/qprocess.html). QProcess only provides `SIGTERM`, via [QProcess::termate()](http://doc.qt.io/qt-5/qprocess.html#terminate) and `SIGKILL` via [QProcess::kill()](http://doc.qt.io/qt-5/qprocess.html#kill).

As `SIGTERM` and `SIGINT` are quite close in semantics according to [this page](https://www.gnu.org/software/libc/manual/html_node/Termination-Signals.html), I think that it would be nice if `rosmon` also handles `SIGTERM`.

What do you think about that?

xqms, 2018-10-26:
I agree, SIGTERM and SIGHUP should be handled gracefully as well. Could you check if #60 fits your requirements?

romainreignier, 2018-10-26:
Yes, it is exactly the patch I have tried locally.
Thank you.

xqms, 2018-10-27:
Merged #60. Thanks for checking!

closed 2018-10-27 by xqms