xqms/rosmon #61: Missing --wait command

romainreignier, 2018-11-06:
`rosmon` is not (yet?) able to start a `roscore` and it is not quite and issue.

But in a multi machines setup, it is quite common that a launch has to wait for the master on a remote computer to come-up.
For now, `rosmon` exits if no master is detected.
I think it would be nice to add a `--wait` option like `roslaunch` to keep the rosmon waiting for the master to come up.

xqms, 2018-11-06:
Sounds like a good idea. Maybe it could even be the default behavior?

I actually don't like auto-starting the roscore, because it often creates confusion ("Will the roscore still be running if I close this roslaunch instance?"). Keeping the user in explicit control of the roscore avoids these issues. That's why I never got around to implementing an auto-start feature and will probably never do it :-)

Anyway, can you think of some existing application that would break if we would simply wait for the roscore instead of exiting? I *think* nobody uses `rosmon` for testing if the roscore is up, so we should be fine...

romainreignier, 2018-11-06:
Yes, sure, having the user to manage rosmaster lifetime sounds better.

And about wait as default. I think it would be nice if there is a clear
message to explain that it is waiting for the rosmaster.

xqms, 2018-11-07:
I merged #62 which implements exactly this.

closed 2018-11-07 by xqms

amilcarlucas, 2020-06-08:
How do I start a roscore in the launch file. This simple information is messing from the README

I know you feel like it should be done manually by the user. But then please add that information to the README

xqms, 2020-06-08:
Hey amilcarlucas,

thanks for your comment. This is a very old issue, it's often better to create a new issue rather than answering to a closed one. I only saw your message by chance.

> How do I start a roscore in the launch file. This simple information is messing from the README

The README points to the ROS wiki page, where this is clearly mentioned (differences to roslaunch). And rosmon prints `roscore is not runnning. Waiting until it is up (abort with CTRL+C)...` during startup to make it clear that nothing is going to happen until someone starts a roscore.

amilcarlucas, 2020-06-08:
My main problem was that by reading the README I got the impression that I could start a roscore by adding a line in the launch file. That is not possible, I now know that. But the information on the README lead me in that direction. IMHO the README can be more explicit regarding that.

xqms, 2020-06-08:
@amilcarlucas Since I'm always interested in improving the documentation, can you be more explicit what exactly you misunderstood in the README?