xqms/rosmon #65: Documentation for rosmon-stop-timeout

alaurenzi, 2018-12-18:
Hi Max,
I had an issue with some ros nodes that take a considerable time to exit (they write a lot of data to disk during shutdown). By default, they are sigkilled by the rosmon after a small timeout duration.
Searching in the code, I found out exactly what I need (`rosmon-stop-timeout`). However, this nice feature is not documented on the ros package page, unless I overlooked it!

xqms, 2018-12-19:
Arturo, thanks for the report! That definitely should have gone into the Wiki. I added it here: https://wiki.ros.org/rosmon/XML. Did you find that page during your search or is it too hidden?

alaurenzi, 2018-12-19:
It’s fine! The xml page was not immediate to find, but if one reads carefully, it’s there

xqms, 2018-12-19:
Thanks for the confirmation, I'll leave it like that then.

closed 2018-12-19 by xqms