xqms/rosmon #66: Loading non-existent text file to param server gives no error

alaurenzi, 2018-12-23:
Hi Max, 
I noticed that loading a non-existent text file to the parameter server does not give any error with rosmon (it does with roslaunch).
A simple example is the following `broken_launch.launch`:
```xml
<launch>

    <param name="myparam" textfile="/my/fake/file"/>

</launch>
```
On my machine (ubuntu 16.04, ros kinetic) rosmon output is:
```
arturo@alaurenzi-iit-laptop ~/Code/ifstream_test $ mon launch broken_launch.launch 
Loaded launch file in 0.000039s
ROS_MASTER_URI: 'http://localhost:11311'
roscore is already running.
Running as '/rosmon_1545570088481348458'

No ROS nodes to be launched. Finished...
```

Looking at the relevant code, the reason appears to be that file read errors are tested with `ifstream::bad()`. However, from the simple test below it looks like the correct way is actually to use `ifstream::operator bool`.

```c++
#include <fstream>

int main()
{
    std::ifstream file("my/fake/file");
    
    if(file.bad())
    {
        throw std::runtime_error("bad");
    }
    
    if(!file)
    {
        throw std::runtime_error("!file");
    }
    
    return 0;
}
```

On my system, the output is
```
arturo@alaurenzi-iit-laptop ~/Code/ifstream_test $ ./ifstream_test 
terminate called after throwing an instance of 'std::runtime_error'
  what():  !file
Aborted (core dumped)
```
which shows that `file.bad()` returns `false`.

xqms, 2018-12-27:
Arturo, thanks for the detailed analysis! I pushed a fix in #68.

xqms, 2018-12-27:
Merged #68.

closed 2018-12-27 by xqms