xqms/rosmon #70: Create a separate package for the rqt plugin

romainreignier, 2019-01-29:
As mentioned in #51 it would be nice a have a separate binary package for the rqt plugin.
This would allow to reduce the dependencies of `rosmon` when used in a robot without GUI.

Such a separation exists in other ROS packages like [grid_map](https://github.com/ANYbotics/grid_map).

I know that this a breaking change, users will lose the plugin when updating, but it may be possible to keep `ros-*-rosmon` name for the metapackage that install both the `rosmon` utility and the rqt plugin.

xqms, 2019-02-03:
Yep, definitely a good idea. My plan would be to go for four packages:

* `ros-*-rosmon-msgs` for message definitions
* `ros-*-rosmon-core` for `rosmon` itself, depending on `msgs`
* `ros-*-rosmon-gui` for the GUI, depending on `msgs`
* `ros-*-rosmon` for a metapackage depending on `core` and `gui`

@romainreignier does that fit with your use case?

romainreignier, 2019-02-03:
Yes, that seems a great séparation.

xqms, 2019-02-03:
On second thought, I'll probably keep the core executable inside the `rosmon` package, since that could also break people's setup (think `rosrun rosmon rosmon ...`).

So the new plan would be:

* `rosmon_msgs` for message definitions
* `rosmon_gui` for the GUI, depending on `rosmon_msgs`
* `rosmon` for `rosmon` itself, depending on `rosmon_msgs`

xqms, 2019-02-03:
I started a PR in #72.

romainreignier, 2019-02-04:
Ah yes, indeed, we must keep the `rosrun rosmon rosmon`. Nice catch!

xqms, 2019-03-11:
#72 is merged and I'm running a prerelease now to see if there are any packaging issues. If that works, I'll perform a release into melodic.

xqms, 2019-03-11:
prerelease test passes, so I created a `2.0.0` release. PR for melodic is open here: https://github.com/ros/rosdistro/pull/20538. As soon as that is merged, we can test the binary packages from the shadow repo.

xqms, 2019-03-12:
First try failed because of a leftover `pluginlib` dependency. Strange, I didn't see that in the prerelease test... Anyway, new PR for 2.0.1: https://github.com/ros/rosdistro/pull/20542

xqms, 2019-03-13:
After another iteration we have binary packages:

* http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-melodic-rosmon-core/ros-melodic-rosmon-core_2.0.2-0bionic.20190313.012143_amd64.deb
* http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-melodic-rosmon-msgs/ros-melodic-rosmon-msgs_2.0.2-0bionic.20190313.011658_amd64.deb
* http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-melodic-rosmon/ros-melodic-rosmon_2.0.2-0bionic.20190313.012815_amd64.deb
* http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-melodic-rqt-rosmon/ros-melodic-rqt-rosmon_2.0.2-0bionic.20190313.012156_amd64.deb

In my test setup everything seems to work.

@romainreignier I'd appreciate it if you could test these on your system (just `sudo dpkg -i *.deb`). Make sure that your workspace does not contain the rosmon sources, otherwise they will shadow the system installation.

romainreignier, 2019-03-13:
Ok, I will test them

xqms, 2019-04-11:
The binary packages have been synced to the master repositories. I actually had some problems during the update due to installation order: `rosmon_core` was installed before the old `rosmon` had been updated - which failed because both packages contained the rosmon libraries. It was solved by triggering the update again, however.

I'm not certain if this kind of problem can be avoided - we would need to specify something like `Conflicts: rosmon <= X.Y.Z` in the debian packages.

Anyway, I'm closing this since the packages have been released. Thanks for your assistance, @romainreignier :-)

closed 2019-04-11 by xqms