xqms/rosmon #71: Is there any way to start new node after rosmon has been executed with launch file?

ddke95, 2019-02-01:
Hi.
Is there any way to start new node after using rosmon after rosmon has been exectued with launch file?
What I'm trying to do is I want to execute rosmon with launch file which has minimum node to boot up a robot right after ubuntu is booted up using desktop autostart.
After this process is finished, which means there are roscore and some nodes are running with rosmon on my robot, I want to run more node using rosmon. The reason why I'm trying to use rosmon is the monitoring function is needed.
Is there any way to start new node after ronmon is once executed?
or Is it possible that I listed up all the nodes that I need on the launch file which will be run while booting process and some of nodes are to start and some are not to start?

xqms, 2019-02-01:
Hi, I'm not sure I understand what you want. Maybe these are options for you?

1) Start `rosmon` using the `--no-start` option. This way, nothing will be started and you can start individual nodes using the ROS service call API or the `rqt` gui, or

2) split the launch file into two, one with the essentials that you'd like to start on boot, and one that you will start later, or

3) implement support for a `rosmon-start="true"/"false"` attribute yourself, which, if set to `false`, would tell rosmon to not start a node automatically. A pull request would be welcome!

ddke95, 2019-02-07:
Thanks for your reply.
Right after I wrote this question, I found the number 1 option that you suggested and tested. number 1 option is good with my intent.
But Can I ask you more about number2, and number 3 that you suggested?
1. Can I launch another launch file using rosmon while one launch file is executed with rosmon?
2. number 3 that you suggested means that "I" implement the function and "I" make pull request, right?

xqms, 2019-02-07:
> Can I launch another launch file using rosmon while one launch file is executed with rosmon?

I don't see why not - open a new shell and run rosmon with another launch file.

> number 3 that you suggested means that "I" implement the function and "I" make pull request, right?

Yes. I'm not particularly interested in that feature, so I won't work on that. If you put in the work I'll be happy to merge it, though.

I'll close this issue since you found a solution - if there a still open problems feel free to reopen it.

closed 2019-02-07 by xqms