xqms/rosmon #81: mon launch crashed No such process

lbxxdi, 2019-05-21:
mon launch crashed after long run, it's happen 2-3 times within 15 hours. all node inside the mon died.

the error 
```
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::status: No such process: "/proc/1301/stat"
```

I run ~10 mons with ~20 nodes on PCM-3365 with Intel Atom CPU and 16 GB RAM. and this PC connect to many sensors.

xqms, 2019-05-21:
Oh, that's definitely a bug - thanks for the report :+1:

I suppose there is a race condition in the process monitor. Is it possible that the node in question (PID 1301) exited just before the rosmon crash?

Do you have a full log available? (a bit more context above the error message probably suffices)

We check existence of the `stat` file here: https://github.com/xqms/rosmon/blob/349a85b6c2803460a2baf4c61ef5747bacfc5c45/rosmon_core/src/monitor/monitor.cpp#L187-L192

But if the process exits in between, we get an unhandled exception in `readStatFile()`.

xqms, 2019-05-21:
It's more strange than I thought - the exception probably originates within the `fs::exists()` call. `readStatFile()` handles I/O errors properly. I guess we can handle failures of `fs::exists()` - I'm just interested why we would get `ESRCH` (No such process) instead of `ENOENT` (No such file or directory). In my tests so far, I only got `ENOENT` error codes from `/proc` files.

xqms, 2019-05-21:
If you have time, I would be grateful if you could test #82 by compiling `rosmon` from source from the `bugfix/remove_stat_check` branch.

lbxxdi, 2019-05-21:
I don't sure about the process (PID 1301) is exited before the rosmon crash. I will continue tomorrow. Thank you so much.

lbxxdi, 2019-05-22:
Already test from #82. It's work !!

Make it happen again by create a lot of short-time processes. 
```for i in `seq 1 100000`; do sleep 2."$RANDOM" & done```
run above command in many terminals. and try to open many mon too.

Here is full error message.
[image]

xqms, 2019-05-22:
Thanks for testing. I tried to reproduce locally using your instructions, but never saw the problem...

Anyway since #82 fixes the problem for you, I'll merge that and prepare a new release.

PR #82 (2019-06-19): rosmon_core: monitor: remove `exists` check for /proc/.../stat
We use boost::filesystem to check if the stat file exists. It seems that
`fs::exists` handles ESRCH (which procfs may return in some cases, see #81)
as an exception, which is then unhandled by us.

Instead of catching the exception, we can just remove the check since
`readStatFile()` handles any error on opening the stat file correctly and
causes a `continue` in the process iteration loop.

This should fix #81.
closed 2019-06-19 by xqms