xqms/rosmon #84: Caught python exception while evaluating $(eval name == true)

matawed, 2019-05-25:
#### problem 

mon launch reports `Caught Python exception while evaluating $(eval parameter_name == true):`

#### input

```
<launch>

    <arg name="arg_name" default="false"/>

    <node name="node_name" pkg="pkg_name" type="node_type" output="screen">
    
        <rosparam if="$(eval arg_name == true)" param="another_param">false</rosparam>
    </node>
</launch>
```
#### workaround
replace `$(eval arg_name == true)` with `$(eval arg_name == True)`

xqms, 2019-05-27:
Fixed in #86. If you like, you could test that from branch `bugfix/python_true_false`.

We'll have to wait for the ROS CI server to be available again before merging.

matawed, 2019-05-30:
Thanks, fix works

PR #86 (2019-06-18): Handle lowercase true/false in $(eval) expressions
For compatibility with `roslaunch`.

Fixes #84.
closed 2019-06-18 by xqms