xqms/rosmon #87: Can't upgrade to version 2.0.2

MichaelT64, 2019-06-20:
Hi
I have the version 1.0.10 and want to upgrade to version 2.0.2.
Following the installation from the ROS wiki I get the following message:

> ros-kinetic-rosmon is already the newest version (1.0.10-0xenial-20190609-214009-0800).
From my understanding of [ROS Index](https://index.ros.org/p/rosmon/github-xqms-rosmon/) I would expect that the new version is also available for kinetic.

xqms, 2019-06-21:
Hey, thanks for the report. Indeed I did not release version 2.x for kinetic so far. I will do so in the next release cycle (probably next week). I currently have the problem that the CI pipeline is not working due to the recent security issues with the ROS buildfarm (https://discourse.ros.org/t/security-issue-on-ros-build-farm/9342).

I don't know how ROS Index works, but here are the repository status pages for kinetic and melodic, which report the version numbers accurately:

http://repositories.ros.org/status_page/ros_kinetic_default.html?q=rosmon
http://repositories.ros.org/status_page/ros_melodic_default.html?q=rosmon

MichaelT64, 2019-06-21:
Hi
thanks, I did not know the  repository status pages (yet), but they seems to be of interest.
Is there a link to it from the ros.org or the wiki to it?

xqms, 2019-06-21:
Well, this is more of a detail for package developers. It's mentioned in Dirk Thomas' presentation of the ROS buildfarm:
http://www.dirk-thomas.net/slides_roscon2016_ros-buildfarm/2016-10-09_roscon_ros-buildfarm.html#/5/6

If you want to improve the situation, consider raising an issue with ROS Index (they should maybe indicate the currently released version): https://github.com/ros-infrastructure/rosindex

xqms, 2019-06-25:
I just released version 2.1.0 for melodic. As soon as the [kinetic sync hold](https://discourse.ros.org/t/preparing-for-kinetic-sync-2019-06-24/9663) is over, I'll perform a release for kinetic as well.

Would you be willing to test 2.1.0 on kinetic? I don't have any kinetic users here anymore, so I only have the CI tests to verify that we didn't break anything. An actual end-user test would be helpful ;-)

You could either compile rosmon yourself, or wait until the build farm has built Ubuntu packages (probably end of this week).

MichaelT64, 2019-06-28:
Hi, I cloned rosmon and build it (after cloning rosfmt too) at the notebook and it seems to work.
But I have only simple launch files (I am just a hobbyist who plays a little bit with my robot and ROS in my free time), so I am not sure if everything works well.
It seems my Raspberry is overextended with building so I have to wait for the build farm.

Greetings
Michael

matawed, 2019-07-02:
> I just released version 2.1.0 for melodic. 

When do you think will it be available in bionic ros repos, since [status page](http://repositories.ros.org/status_page/ros_melodic_default.html?q=rosmon) still reports a difference between `testing` and `main` for rosmon? 
[Package index](http://repositories.ros.org/ubuntu/main/dists/bionic/main/binary-amd64/Packages) for bionic still only provides 2.0.2.

xqms, 2019-07-02:
When the next repo [sync](https://discourse.ros.org/t/preparing-for-melodic-sync-2019-06-28/9714) happens. This is done usually every 2-4 weeks and is manually initiated by the ROS maintainers. You can monitor the sync announcements here:
https://discourse.ros.org/c/release

xqms, 2019-07-02:
In the meantime, if you want to beta-test (I'm always happy if we get more people doing this), you can switch your system to the shadow repository, which always contains the newest packages released by the package maintainers (i.e. me in this case):
http://wiki.ros.org/ShadowRepository

MichaelT64, 2019-07-03:
At least I get catkin_make to build the sources at my Raspberry.
Everything seems to work (at least at my simple launch files).

xqms, 2019-07-03:
Awesome, thanks for testing @MichaelT64 !

I'll trigger the release into Kinetic tomorrow.

xqms, 2019-07-05:
Release PR for kinetic: https://github.com/ros/rosdistro/pull/21705
Will close this as soon as the PR is merged.

xqms, 2019-07-12:
Packages for kinetic are out in the repositories.

closed 2019-07-12 by xqms