xqms/rosmon #90: Timestamp in console output

MichaelT64, 2019-07-03:
Is it possible to add the timestamp of the message in the console output?
Than you can see not only from which node a message comes but also when it comes (and maybe how often).
The printing of the timestamp can be controlled by an option.

xqms, 2019-07-04:
You already have two options here:

1. Look in the log files (by default, rosmon produces logfiles in `/tmp/rosmon*.log`), they are timestamped. It's best to view them with `less -R`, that way the message colors do not get lost.
2. `export ROSCONSOLE_FORMAT='[${severity}] [${time}]: ${message}'` or similar before launching rosmon. That way you will get timestamps in the terminal UI. This affects only output through rosconsole, though (`ROS_INFO()` and friends). Plain `std::cout` or `printf` will not be timestamped.
More information on that here: http://wiki.ros.org/rosconsole

In my experience I usually don't look at timestamps *while* the system is running, but rather afterwards, so the log file should suffice and an additional option for the terminal UI is not required. What do you think?

xqms, 2019-07-12:
I'm closing this now - if you feel the issue is not resolved, feel free to reply.

closed 2019-07-12 by xqms