xqms/rosmon #91: [rosrun] Couldn't find executable named _shim below /opt/ros/melodic/share/rosmon_core

matawed, 2019-07-09:
After upgrading to 2.1.0-1 on Ubuntu 18.04 and launching a previously running launch file now fails. 
All nodes crash with the following error message:

```
[rosrun] Couldn't find executable named _shim below /opt/ros/melodic/share/rosmon_core
node: node exited with status 3
```

whereas the same launch file runs fine with roslaunch.

Was there some error during packaging? Seems that [this](https://github.com/xqms/rosmon/blob/426c0772a8324b5bccd76135adcb5494887f2918/rosmon_core/CMakeLists.txt#L73) executable is missing.

`dpkg -L ros-melodic-rosmon-core`  yields:
```
.
/opt
/opt/ros
/opt/ros/melodic
/opt/ros/melodic/etc
/opt/ros/melodic/etc/catkin
/opt/ros/melodic/etc/catkin/profile.d
/opt/ros/melodic/etc/catkin/profile.d/50-rosmon.bash
/opt/ros/melodic/etc/catkin/profile.d/50-rosmon.zsh
/opt/ros/melodic/lib
/opt/ros/melodic/lib/librosmon_launch_config.so
/opt/ros/melodic/lib/pkgconfig
/opt/ros/melodic/lib/pkgconfig/rosmon_core.pc
/opt/ros/melodic/lib/rosmon_core
/opt/ros/melodic/lib/rosmon_core/rosmon
/opt/ros/melodic/share
/opt/ros/melodic/share/rosmon_core
/opt/ros/melodic/share/rosmon_core/catkin_env_hook
/opt/ros/melodic/share/rosmon_core/catkin_env_hook/50-rosmon.bash
/opt/ros/melodic/share/rosmon_core/catkin_env_hook/50-rosmon.zsh
/opt/ros/melodic/share/rosmon_core/cmake
/opt/ros/melodic/share/rosmon_core/cmake/rosmon_coreConfig-version.cmake
/opt/ros/melodic/share/rosmon_core/cmake/rosmon_coreConfig.cmake
/opt/ros/melodic/share/rosmon_core/package.xml
/usr
/usr/share
/usr/share/doc
/usr/share/doc/ros-melodic-rosmon-core
/usr/share/doc/ros-melodic-rosmon-core/changelog.Debian.gz
```

xqms, 2019-07-09:
Urgh, that was not supposed to happen (and I was supposed to test the built package before the melodic sync happened -.-). Thanks for reporting!

I'll push a new patch release shortly.

xqms, 2019-07-09:
This is one of the possible mistakes that currently slip through the CI pipeline and requires manual testing of the binary packages that are built by the ROS buildfarm. Somehow I messed up my timing this release cycle and didn't do that step. I apologize to everyone affected.

Release PRs for Kinetic and Melodic:
https://github.com/ros/rosdistro/pull/21724
https://github.com/ros/rosdistro/pull/21723

If you need a fix/workaround now, you can:
 * compile rosmon from source (the fix is in master), or
 * wait for the relevant PR to be merged and the buildfarm to build packages, then install these manually (I'll post instructions later), or
 * possibly downgrade to version 2.0.1, which does not have this problem. If you are lucky, your local apt cache still has this version (try `apt-cache policy ros-melodic-rosmon-core`)

romainreignier, 2019-07-17:
@xqms Where could we find the packages built by the buildfarm?
I am not very familiar with the ROS buildfarm.
Thank you.

xqms, 2019-07-18:
@romainreignier You can either switch your apt sources to the "shadow" repository, which will automatically give you the released versions even before the sync happens.
I would not recommend this on your usual developer or even production machine.
Although using it *greatly* helps package maintainers because hopefully bugs like this one can be caught before they hit the official repo.

http://wiki.ros.org/ShadowRepository

The other possibility is to download the rosmon packages manually from the repository, and install/update only these:

http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-melodic-rosmon-core/ros-melodic-rosmon-core_2.1.1-1bionic.20190709.214808_amd64.deb
http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-melodic-rosmon-msgs/ros-melodic-rosmon-msgs_2.1.1-1bionic.20190709.213006_amd64.deb
http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-melodic-rosmon/ros-melodic-rosmon_2.1.1-1bionic.20190709.215647_amd64.deb
http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-melodic-rqt-rosmon/ros-melodic-rqt-rosmon_2.1.1-1bionic.20190709.214533_amd64.deb

Install the downloaded `.deb` files all *at once* with `sudo dpkg -i <files...>`.

This procedure is only required for ROS Melodic, ROS Kinetic already has the fixed version synced.

xqms, 2019-10-05:
The fixed versions have been in the repositories for quite a while now, closing.

closed 2019-10-05 by xqms