xqms/rosmon #92: zsh completion of packages does not work

matawed, 2019-07-09:
If I enter 
`mon launch <first_letters_of_package_name><TAB><TAB>`
in a sourced workspace, then I get the following:

```
(eval):1: no matches found: *.launch:globbed-files
(eval):1: no matches found: *.launch:globbed-files
(eval):1: no matches found: *.launch:globbed-files
```

Bash completion works.

xqms, 2019-07-09:
Hm, I don't use zsh myself. I doubt that there has been a regression (the zsh completion code has remained largely unchanged), probably you are hitting a problem that was always there.

If you are interested in this, feel free to look into it - as mentioned, I'm not a zsh expert myself...

matawed, 2019-07-10:
Yes, most likely this is no degradation. 

I'll have a look at it. One of the problems is in this [line](https://github.com/xqms/rosmon/blob/3f41b1324dce62f0b2b29f6c52f1d0f3b647d5ec/rosmon_core/env-hooks/50-rosmon.zsh#L25) if there is no launch file in this directory globbing fails and thus this error is reported.

d-walsh, 2020-05-01:
### 1)  `*.launch:globbed-files` Error
I think one of the problems is when you source using the standard `setup.zsh` it doesn't load/use the environment hooks correctly:
```
➜  source /opt/ros/melodic/setup.zsh 
➜  mon launch 
(eval):1: no matches found: *.launch:globbed-files
```
A workaround is to directly source the rosmon environment hooks:
```
➜  source /opt/ros/melodic/etc/catkin/profile.d/50-rosmon.zsh
```

This avoids the `*.launch:globbed-files` error.

### 2)  `_values:compvalues:11: not enough arguments` Error
There is another error:
```
➜  mon launch image_geometry 
_values:compvalues:11: not enough arguments
_values:compvalues:11: not enough arguments
_values:compvalues:11: not enough arguments
```

And I have a fix for it - it's because even with an empty string the size of files is 1.
In https://github.com/xqms/rosmon/blob/master/rosmon_core/env-hooks/50-rosmon.zsh#L47
```
		if [[ $#files -gt 0 ]] && [[ -n $files ]]; then
			_values 'launch files' $files
		fi
```

xqms, 2020-05-02:
Hey, thanks for the information :)

Regarding 1): As far as I know the `setup.zsh` should do exactly the same as directly sourcing our file... Maybe it's an ordering issue? Maybe we get loaded too early in `setup.zsh`?

Regarding 2): Nice! Maybe we can prepare a PR to collect the fixes. I don't have time to work on this right now, though (and zsh is pretty low on my priority list, as I don't use it :-P).