xqms/rosmon #93: Support of more than 62 nodes

matawed, 2019-07-22:
Unfortunately, if I start more than 62 nodes, then I don't have the possibility to control the nodes >62 in the command line interface as I can only refer to nodes 0..61 via a-z,A-Z,0-9. 

Did you ever think of an command line interface for more than 62 nodes?

xqms, 2019-07-24:
I've been thinking about a search mode for a while now (like less/vi, triggered with '/'), which would reduce the list of nodes to a number that is manageable using the current shortcut system.

Do you have a better/different idea?

matawed, 2019-07-25:
I like the idea with the search mode similar to vim.

clungzta, 2019-10-16:
Search would be great! Another benefit of the search is less time and difficulty to perform actions on a node. 

For example: as a new user, if I wish to perform an action to my `move_base` node, I instinctively to type 'm' when in fact it is assigned node key 'a', then I have to look at the list and correct it.

clungzta, 2019-10-16:
Variant b sounds quite nice! :+1:

xqms, 2019-10-16:
I implemented a draft solution in #97. Let's continue the discussion there. Beta-testers welcome!

xqms, 2019-10-17:
Fixed in #97.

closed 2019-10-17 by xqms