xqms/rosmon #94: loading nonexist yaml file for rosparam doesn't cause failure

mikexing2018, 2019-08-09:
if my launch xml file has something like <rosparam command='load' file='abc.yaml'>, rosmon doesn't fail if abc.yaml doesn't exist.  Roslaunch seems to fail and throw an error. Is there a reason for this difference in behavior?

xqms, 2019-08-09:
Hm, in my case I get an error:

```
<launch>
        <rosparam command="load" file="/tmp/not_there.yaml" />
</launch>
```

leads to:

```
$ mon launch test.launch 
Could not load launch file: test.launch:2: Could not open file '/tmp/not_there.yaml'
```

Which version of rosmon are you using? Compiled from source or repository packages?
Could you share a minimum example of a full launch file demonstrating your problem?

xqms, 2019-10-05:
Do you have any further details on this issue? Otherwise I'll close it since I cannot reproduce it.

xqms, 2019-10-07:
I'll close this for now. If you have more details, feel free to reply and I'll reopen the issue.

closed 2019-10-07 by xqms

mikexing2018, 2019-10-07:
Yes. Closing it is fine. THanks.