xqms/rosmon #95: endStateUpdate() is taking a hardcoded value assuming update every 1 second

mikexing2018, 2019-10-04:
In monitor.cpp, endStatUpdate() should take an elapsed time argument. Instead, it's currently taking in the return value of kernel_hz(), which is just a hardcoded number of ticks per second...

xqms, 2019-10-05:
Yes, the code assumes that `Monitor::updateStats()` is called every second - but the timer should ensure that with high precision. Did you notice any deviation?

Division by `KERNEL_HZ` is necessary since the kernel reports the CPU times in tick counts.

mikexing2018, 2019-10-05:
Right, with assumption of every 1 second, updateStats() worked fine. It was when I changed the interval to something other than 1 second that it started to give out the wrong stats for CPU because of the underlying assumption. The code could be improved to handle calling updateStats() every n seconds.

xqms, 2019-10-05:
Fixed in febb13cd40ef5f029fab387d7040d808059a4d3f. I think there is no reason for a new binary release, since this does not affect users.

closed 2019-10-05 by xqms

mikexing2018, 2019-10-07:
@xqms, thanks for the quick responses as always.

commit febb13cd (2019-10-05): rosmon_core: monitor: make updateStats() rate-independent

This fixes #95.