xqms/rosmon #100: Handling of "anon" names not consistent with roslaunch

marco-tranzatto, 2019-10-31:
Hi all,

once more thanks for sharing this great tool.

I have been using it a lot lately and found a behavior mismatch with roslaunch about the usage of "anon" names for nodes.

I created a minimal example [here](https://github.com/marco-tranzatto/launch_file_examples/tree/master/rosmon_anon_issue).
The package contains two launch files: 

- "anon_node_include.launch" which is a simple talker node whose name is `name="$(anon talker)"`
- "main.launch" which includes the file "anon_node_include.launch" two times.

If the user executes
```
roslaunch rosmon_anon_issue main.launch
```
Then the launch succeeds and there are two nodes with anonymous names:
```
$ rosnode list 
/rosout
/talker_rsl_l25_4797_5574635845418076715
/talker_rsl_l25_4797_7425401142298951921
```

On the other hand, if the user types
```
mon launch rosmon_anon_issue main.launch
```

Then the following error is reported: 
```
Could not load launch file: /home/marcot/catkin_ws/src/launch_file_examples/rosmon_anon_issue/launch/anon_node_include.launch:6: node name 'talker_52666652' is not unique
```

@xqms any hints about this?

Thanks a lot for your help,
Marco.

xqms, 2019-10-31:
That definitely sounds like a bug. Thanks for providing an example, that'll help in creating a test for it.

When I implemented `$(anon XYZ)`, I thought that the same `XYZ` prefix should always resolve to the same node name. But it seems that this is only true at file scope, as demonstrated by your multiple-inclusion example.

I'll have time to look at this in more detail next week.

xqms, 2019-11-04:
I found a smaller reproducing example, which shows some more interesting behavior:

```xml
<launch>
  <param name="test_1_global" value="$(anon test_1)" />

  <group>
    <param name="test_1_local" value="$(anon test_1)" />
    <param name="test_2_local" value="$(anon test_2)" />
  </group>

  <param name="test_2_global" value="$(anon test_2)" />
</launch>
```

In this case, we get from `roslaunch`:
```
test_1_global: test_1_martin_11923_1971446669036573556
test_1_local:  test_1_martin_11923_1971446669036573556 (same)

test_2_global: test_2_martin_11923_7236432865625322132
test_2_local:  test_2_martin_11923_5520981918244494059 (different)
```

It seems the effect of `$(anon)` is local, but the lookup is recursive into the parent scopes. Ordering also matters.

marco-tranzatto, 2019-11-04:
Interesting! I didn't think about param names, since the wiki mentions only for nodes:

> This is used for <node> name attributes in order to create nodes with anonymous names [1]

what's your intention on this issue?

[1] [Wiki](http://wiki.ros.org/roslaunch/XML#substitution_args), see "$(anon name)" section.

xqms, 2019-11-04:
> Interesting! I didn't think about param names, since the wiki mentions only for nodes:

Yes - `<param>` is just easier for me to test in the unit test framework ;-)

> what's your intention on this issue?

I opened a pull request with a fix at #101. It requires a bit more system-level testing, though.

By the way, congrats on opening the 100th issue :tada: :smile:

marco-tranzatto, 2019-11-04:
I see your point about testing.

Never had this record before about the 100th issue xD

Again, thanks for sharing and taking care of this awesome tool!
Marco.

closed 2020-04-10 by marco-tranzatto