xqms/rosmon #105: Could not find "node" in "package" after catkin_make install

nPellejero, 2019-11-14:
Hello,
I am using rosmon, and trying to install my packages with catkin_make install, in /opt/my_custom_install_dir and I am experiencing the following issue:

When launching rosmon, after sourcing the local setup.bash and trying to launch all the nodes, every of them are throwing the same exception: 

node_name:   what():  Could not find node 'node_name' in package '/package_name'

How should I debug this problem? I keep changing the final destination of the binaries, but I am not sure of where is rosmon looking for the binaries, there is a verbosity level or some tool to debug this kind of things? 

Thanks,

Nicolas.

xqms, 2019-11-19:
> node_name: what(): Could not find node 'node_name' in package '/package_name'

That leading `/` in `/package_name` looks suspicious, could you try removing that?

Otherwise, the logic for finding executables is in here:
https://github.com/xqms/rosmon/blob/abdae1f8bb4f8684ea45fb36f4fe7730bcd102ea/rosmon_core/src/package_registry.cpp#L94-L120
You could try debugging that / adding some debug prints here.

Does `rosrun package_name node_name` work?

nPellejero, 2019-11-19:
Hi Xqms, after a few more iterations, I made it to found the right place for the binaries. It seems to be (obviously) sound with the logic in your function. Thanks for the response, it is useful to confirm the exact logic.I think you can close this issue.

Nicolas.

closed 2019-11-19 by xqms