xqms/rosmon #115: Something about bad addresses, signal 6, and non-empty directories

mjsobrep, 2020-04-23:
I have this occasional problem working with rosmon (which is otherwise incredible, thank you) where on launch ~half of my nodes crash. It seems to be a problem from an interaction with rosmon. Sometimes restarting the system solves it, sometimes it does not. I think that rosmon is building directories to keep track of something and not properly cleaning them up / not tolerating messy shutdowns? Any thoughts on how to fix this?

Snippet of the print out after a failed launch:

<pre><span style="background-color:#D3D7CF"><font color="#2E3436">    image_throttled:</font></span> Could not execute /opt/ros/kinetic/lib/topic_tools/throttle messages image_raw 15 image_throttled __name:=image_throttled : Bad address
<span style="background-color:#D3D7CF"><font color="#2E3436">     image_throttled:</font></span> image_throttled died from signal 6
<span style="background-color:#D3D7CF"><font color="#2E3436">     image_throttled:</font></span> image_throttled left a core dump
<span style="background-color:#D3D7CF"><font color="#2E3436">     image_throttled:</font></span> Determined pattern &apos;/tmp/rosmon-node-zKMbN7/core&apos;
<span style="background-color:#D3D7CF"><font color="#2E3436">     image_throttled:</font></span> Found core file &apos;/tmp/rosmon-node-zKMbN7/core&apos;
<span style="background-color:#D3D7CF"><font color="#2E3436">     image_throttled:</font></span> Could not remove process working directory &apos;/tmp/rosmon-node-zKMbN7&apos; after process exit: Directory not empty
<span style="background-color:#D3D7CF"><font color="#2E3436">         mobile_base:</font></span> [loadNodelet]: Loading nodelet /mobile_base of type kobuki_node/KobukiNodelet to manager mobile_base_nodelet_manager with the following remappings:
<span style="background-color:#D3D7CF"><font color="#2E3436">         mobile_base:</font></span> [loadNodelet]: /mobile_base/joint_states -&gt; /joint_states
<span style="background-color:#D3D7CF"><font color="#2E3436">         mobile_base:</font></span> [loadNodelet]: /mobile_base/odom -&gt; /odom
<span style="background-color:#D3D7CF"><font color="#2E3436">         mobile_base:</font></span> [service::exists]: waitForService: Service [/mobile_base_nodelet_manager/load_nodelet] has not been advertised, waiting...
<span style="background-color:#D3D7CF"><font color="#2E3436">   realsense2_camera:</font></span> Could not execute /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera : Bad address
<span style="background-color:#D3D7CF"><font color="#2E3436">realsense2_camera_manager:</font></span> Could not execute /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager : Bad address
<span style="background-color:#D3D7CF"><font color="#2E3436">   realsense2_camera:</font></span> realsense2_camera died from signal 6
<span style="background-color:#D3D7CF"><font color="#2E3436">   realsense2_camera:</font></span> realsense2_camera left a core dump
<span style="background-color:#D3D7CF"><font color="#2E3436">   realsense2_camera:</font></span> Determined pattern &apos;/tmp/rosmon-node-DiS9mQ/core&apos;
<span style="background-color:#D3D7CF"><font color="#2E3436">   realsense2_camera:</font></span> Found core file &apos;/tmp/rosmon-node-DiS9mQ/core&apos;
<span style="background-color:#D3D7CF"><font color="#2E3436">   realsense2_camera:</font></span> Could not remove process working directory &apos;/tmp/rosmon-node-DiS9mQ&apos; after process exit: Directory not empty
<span style="background-color:#D3D7CF"><font color="#2E3436">realsense2_camera_manager:</font></span> realsense2_camera_manager died from signal 6
<span style="background-color:#D3D7CF"><font color="#2E3436">realsense2_camera_manager:</font></span> realsense2_camera_manager left a core dump
<span style="background-color:#D3D7CF"><font color="#2E3436">realsense2_camera_manager:</font></span> Determined pattern &apos;/tmp/rosmon-node-j6bTbd/core&apos;
<span style="background-color:#D3D7CF"><font color="#2E3436">realsense2_camera_manager:</font></span> Found core file &apos;/tmp/rosmon-node-j6bTbd/core&apos;
<span style="background-color:#D3D7CF"><font color="#2E3436">realsense2_camera_manager:</font></span> Could not remove process working directory &apos;/tmp/rosmon-node-j6bTbd&apos; after process exit: Directory not empty
<span style="background-color:#D3D7CF"><font color="#2E3436">          polly_node:</font></span> [start]: polly running: rosrpc://192.168.1.25:42125
<span style="background-color:#D3D7CF"><font color="#2E3436">  keyop_vel_smoother:</font></span> [loadNodelet]: Loading nodelet /keyop_vel_smoother of type yocs_velocity_smoother/VelocitySmootherNodelet to manager mobile_base_nodelet_manager with the following remappings:
<span style="background-color:#D3D7CF"><font color="#2E3436">  keyop_vel_smoother:</font></span> [loadNodelet]: /keyop_vel_smoother/odometry -&gt; /odom
</pre>

xqms, 2020-04-23:
Uhh, that looks like a memory bug, since we get EFAULT ("Bad address") from execvp()...
Could you please specify which ROS + rosmon version you are using? Repository packages or built from source?

I recall that we fixed a nasty bug in #103, maybe this is related.

mjsobrep, 2020-04-23:
This is on Kinetic coming from repository:

<pre>Package: ros-kinetic-rosmon
Status: install ok installed
Priority: extra
Section: misc
Installed-Size: 29
Maintainer: Max Schwarz &lt;max.schwarz@uni-bonn.de&gt;
Architecture: amd64
Version: 2.1.1-1xenial-20200229-053707+0000
Depends: ros-kinetic-rosmon-core, ros-kinetic-rqt-rosmon
</pre>

mjsobrep, 2020-04-23:
A bit more intuition: It seems like all of the nodes that are having trouble are either capturing from cameras or in the same namespace as a camera capturing node. There are other nodes using USB devices that seem ok. This system is often shut down messily (the os is shutdown without regard for what is running).

mjsobrep, 2020-04-23:
Looks the same as #103 which was fixed in v2.2.1. On Kinetic, 2.1.1-1 is the released version. Any chance we can get a bump in released version or should I just build from source?

xqms, 2020-04-24:
Yes, sure. I always delay the Kinetic release a bit to test things on Melodic first, but this time I simply forgot to do the Kinetic update.

I triggered a release right now. The rosdistro PR is here: https://github.com/ros/rosdistro/pull/24555

mjsobrep, 2020-04-24:
Awesome, thanks

xqms, 2020-05-02:
The packages have built, but are not pushed into the main repositories yet. If you like & have time, you could already test them:

 * http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-kinetic-rosmon-core/ros-kinetic-rosmon-core_2.2.1-1xenial-20200427-194103+0000_amd64.deb
 * http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-kinetic-rosmon/ros-kinetic-rosmon_2.2.1-1xenial-20200427-194637+0000_amd64.deb
 * http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-kinetic-rosmon-msgs/ros-kinetic-rosmon-msgs_2.2.1-1xenial-20200427-193637+0000_amd64.deb
 * http://repositories.ros.org/ubuntu/testing/pool/main/r/ros-kinetic-rqt-rosmon/ros-kinetic-rqt-rosmon_2.2.1-1xenial-20200427-194111+0000_amd64.deb

xqms, 2020-05-05:
The package sync is out => rosmon 2.2.1 including the fix for this issue can be installed from the repositories. I'll close this now, feel free to reopen if the issue still persists :)

closed 2020-05-05 by xqms