xqms/rosmon #116: Increase diagnostics thresholds in included launch nodes

romainreignier, 2020-05-14:
Basically, the Gazebo simulator is usually launched by including the `gazebo_ros empty_world.launch` launch file and passing some arguments to it like the `world_name`.

But Gazebo takes a lot of resources so one might want to increase the `rosmon-cpu-limit` on it.

Until now, this attribute can only be used on a `node` element.

So I was wondering if we could allow to use it (and its sibling `rosmon-memory-limit`) to the `include` element and apply it to all nodes found in this included launch file?

PS: The diagnostics thresholds are not documented neither in the [ROS Wiki page](https://wiki.ros.org/rosmon) nor the README.

xqms, 2020-05-24:
We could allow those attributes on all scoping elements (`<launch>`, `<group>`, `<include>`, `<node>`). I'll take a look after I had a chance to go through the open PRs.

Documentation: You are entirely correct, this is missing. This is what happens if I don't use a feature myself ;)

xqms, 2020-05-28:
Fixed in #124 and documented in the ROS wiki.

closed 2020-05-28 by xqms

romainreignier, 2020-05-28:
Thanks. I do not see it on the Wiki page, maybe it takes some time to update?

xqms, 2020-05-28:
Have you looked at https://wiki.ros.org/rosmon/XML? For me it shows the previously undocumented memory & cpu limits and also mentions that these can be applied on `<include>` and friends.

romainreignier, 2020-05-28:
Oh no, sorry, I did not see this page existence before. I was only looking at https://wiki.ros.org/rosmon
Great. Thanks again!