xqms/rosmon #120: Release for Noetic

Timple, 2020-05-22:
The [call for testing](https://discourse.ros.org/t/ros-noetic-call-for-testing) today revealed that this package has not yet been released for noetic. 
Would you mind releasing for noetic as well? :slightly_smiling_face:

xqms, 2020-05-24:
We have two dependencies which have to be released for Noetic first: `catch_ros` and `rosfmt`. I started the release process for those two:
https://github.com/ros/rosdistro/pull/25043
https://github.com/ros/rosdistro/pull/25044

I also did a preliminary check inside a Docker container, and everything seems to compile under Noetic & the unit tests pass.

xqms, 2020-05-28:
I bumped the version to 2.3.0 and triggered a release to Noetic: https://github.com/ros/rosdistro/pull/25139

Timple, 2020-05-28:
Great! Thanks for the follow up.

While you're in releasing mode, can you also re-release for melodic so we can use [this feature](https://github.com/xqms/rosmon/pull/124) right away? :slightly_smiling_face:

xqms, 2020-05-30:
After solving some more release issues, packages for Noetic are in the testing repos:
http://wiki.ros.org/TestingRepository

I'd appreciate if there are people willing to test these in their Noetic setups :)

Release PR for Melodic is open at https://github.com/ros/rosdistro/pull/25210.

closed 2020-06-17 by Timple