xqms/rosmon #126: Release in packages.ros.org depends on rqt which pulls in all of Qt5

rgreid, 2020-06-04:
Great package, thanks for your contribution. We're building a minimal install for a robot using `rosmon`. The standard `ros-melodic-rosmon` release in packages.ros.org depends on `rqt` which pulls in all of Qt5 and a few hundred MB of the Ubuntu desktop environment. This is unnecessary for a robot that's deployed in the field and never runs a desktop.  

Have you any thoughts how to remove the dependency? 

We're working around this at the moment by building from source with `touch rqt_rosmon/CATKIN_IGNORE`.

Timple, 2020-06-04:
This depencency seems wrong: https://github.com/xqms/rosmon/blob/e6ccc7a6364868d02373493adc44fdc531357f0d/rosmon/package.xml#L15

`rqt_rosmon` is not used in this package and is thus not a dependency?

xqms, 2020-06-04:
`rosmon` is a metapackage depending on `rosmon_core` and `rqt_rosmon`. If you don't need the GUI, you can just install `rosmon_core`.

xqms, 2020-06-04:
For discussion on this see also #70.

xqms, 2020-06-04:
Since this is obviously not documented well: Where would you expect to find this? I'll look at modifying the ROS wiki.

Timple, 2020-06-04:
You are right, I was looking at the metapackage. Dependencies are in order!

I think the confusion starts that I can do `rosrun rosmon rosmon` but have to install `rosmon_core` to get this behavior? 

- - -
My personal view: I've never cared much for meta packages. I would remove the metapackage and rename `rosmon_core` to `rosmon`. People who like the rqt interface can install that one.

rgreid, 2020-06-04:
Thanks for clarifying.  @xqms: the first place I'd look for documentation would be `README.md` in this repo.  Second place I'd look is the [ROS wiki](https://wiki.ros.org/rosmon)-- second place because the wiki is often less well maintained. 

Perhaps consider merging some of the ROS wiki text back into `README.md`, and/or adding a quick description for each of the `rosmon` packages.

In my case earlier today, I did actually check `ros-melodic-rosmon`:
```bash
robot@robot:~$ dpkg -L ros-melodic-rosmon 
...
/opt/ros/melodic/share/rosmon
```
Which I'd incorrectly assumed was the `rosmon` binary!  I'll swap over to 
```
apt-get install ros-melodic-rosmon-core
```
Thanks!

closed 2020-06-04 by rgreid

xqms, 2020-06-04:
I edited the installation instructions in both the ROS wiki and the README.md to make this clear.

> I would remove the metapackage and rename rosmon_core to rosmon. People who like the rqt interface can install that one.

Yep, in an ideal world I'd do that as well - but it would break existing installations where people installed `rosmon` only and expect to have the rqt plugin available. That's why we came up with the core/rqt split and a metapackage called `rosmon`.