xqms/rosmon #128: Using terminal ui when running rosmon and non-rosmon nodes in same terminal

timonh, 2020-06-08:
I am planning to run a set of nodes with rosmon while others run with roslaunch. (all in one terminal; running amongst the other nodes a rosmon node with a launch file as argument)

I observed, that the combination of console ouput from roslaunch nodes and the rosmon terminal ui tends to clutter the console output.

Is there a recommended way to use the terminal ui in this case?

xqms, 2020-06-08:
I assume that you are running rosmon nested inside roslaunch.

In that case, rosmon has no control over what and when other nodes (or roslaunch) print to the terminal. Maintaining the terminal UI properly is impossible in this case. There is no way to now in what state the terminal is going to be when rosmon wants to redraw its UI.
Your only choice is to switch it off using the `--disable-ui` option.

Can I ask what your use case is? Why don't you run everything inside rosmon?

timonh, 2020-06-08:
Thanks a lot for your quick response, this is very helpful to know.

We are relying on a launching structure which mainly uses the roslaunch python API. (We need rosmon with certain nodes for a specific application, but most other parties that use this launching structure with roslaunch.)

For this I considered the nested approach to be most resilient to issues that may arise when incorporating third party changes.

closed 2020-06-08 by timonh