xqms/rosmon #130: YAML params in global namespace are not respected

jarvisschultz, 2020-07-22:
Imagine I have the following files:

**example.launch:**
```xml
<launch>
  <group ns="example">
    <rosparam command="load" file="/path/to/test.yaml" />
  </group>
</launch>
```
**test.yaml:**
```yaml
/fake_param: 20.0
```
Using `roslaunch`, this ends up with a parameter of `/fake_param`, using `rosmon` instead ends up with a parameter of `/example/fake_param`. Reading ROS docs on YAML restrictions, I can't find anywhere where globally resolved names in YAML files are described in a REP. So it's not clear if one _should_ do that or what the expected outcome would be. Regardless, the behavior between `rosmon` and `roslaunch` is not the same.

xqms, 2020-07-23:
Yep, that's a bug (like any other non-conformance with `roslaunch`). Thanks for reporting and condensing it down to a minimal example :+1:

> Reading ROS docs on YAML restrictions, I can't find anywhere where globally resolved names in YAML files are described in a REP.

I'm not surprised, a lot of `roslaunch`'s behavior does not appear in any spec :shrug:

There's a tentative fix in #131.

xqms, 2020-07-23:
While setting up a test case I found some even crazier cases:
```
<rosparam ns="ns">
/fake_param: 20.0
namespace:
  /other_param: 10.0
  /global_ns:
    third_param: 5.0
</rosparam>
```
https://github.com/xqms/rosmon/blob/master/rosmon_core/test/xml/test_rosparam.cpp#L264-L270

Anyway, tests pass and roslaunch has the same behavior on the test cases, so I merged #131. Thanks again for reporting!

closed 2020-07-23 by xqms

jarvisschultz, 2020-07-23:
Awesome... thanks for the fast response! Fixes the issue I was seeing.