xqms/rosmon #132: Disable/reduce log data

d-walsh, 2020-08-16:
I am using rosmon on a machine that doesn't get reboot often (clearing tmp) and the nodes that are managed by rosmon can run for days. This leads to very large amounts of data being written to `/tmp/rosmon*` until I eventually get the error:
```
terminate called after throwing an instance of 'fmt::v6::system_error'
  what():  cannot write to file: No space left on device
```
It would be good to be able to disable logging. Or only log recent data (maybe 1 hr?). Also it would be good to handle this error where there is no space available.

xqms, 2020-08-19:
Thanks for the report!

I guess you can already disable logging through `mon launch --log=/dev/null ...`. Do you think we need an explicit option for this?

For log rotation (e.g. keep the last day of logs), you can use an external helper like `logrotate`.

What do you think would be a good recovery behavior if the disk is full?

amilcarlucas, 2020-09-14:
@xqms can you provide more details about the logrotate solution? What copy mode should be used?

amilcarlucas, 2020-09-21:
Would be nice to get an answer :(

xqms, 2020-09-21:
@amilcarlucas sorry, didn't get around to replying yet.

Well, the solution obviously depends on your application... One configuration I can think of is using `--log /path/file` to force rosmon to write its logfile to a particular location. Then use `logrotate` with its `copytruncate` option to periodically "cut" the logfile into segments. The usual options can then be used to delete old log file segments.

amilcarlucas, 2021-08-03:
@xqms nope, that does not work.
I used your idea and this configuration
```
copytruncate
nocompress
notifempty
olddir /media/ubuntu/sdcard/logs/2021-07-05-1409
/tmp/iav-*.log { }
/media/ubuntu/sdcard/.ros/log/*.log { }
/media/ubuntu/sdcard/.ros/log/latest/*.log { }
```

But the log file generated by rosmon still has a ton of zeros in the beginning of the file. Looks like the fseek is somehwere inside the file and so a new file gets created with a bunch of zeros.
The file size increases, instead of going back to zero.
Any other ideas?

amilcarlucas, 2021-08-04:
I got some more info:

- [Problem explanation](https://serverfault.com/questions/221337/logrotate-successful-original-file-goes-back-to-original-size)
- [Possible solution](https://stackoverflow.com/questions/462122/detecting-that-log-file-has-been-deleted-or-truncated-on-posix-systems)

amilcarlucas, 2021-08-16:
I try to address this on https://github.com/xqms/rosmon/pull/154

xqms, 2023-05-05:
rosmon now supports `--log=syslog`, which logs to the systemd journal under Ubuntu. The journal has its own settings for log retention and is a superior solution to logrotate & friends. I'll therefore close this issue. Feel free to comment further if there is still a problem.

closed 2023-05-05 by xqms