xqms/rosmon #134: Terminal width issue with GNU screen

zlacelle, 2020-10-28:
Hello,
We're running all of our processes inside of GNU screen, which is a useful way to allow multiple remote logins to the same terminal as well as background processes. The issue is that the terminal width when not attached looks to be 80 characters, and screen dynamically changes the width once you attach. However, looks like rosmon leaves the old printouts at 80 characters (new entries correctly print at the new width).

Is there a way to get rosmon to update the printed items width dynamically? If I use roslaunch, gnuscreen seems to handle this just fine.

xqms, 2020-10-28:
Hey, thanks for your question ;)

I think this is a limitation of rosmon's current design. Since we want to prefix each line with the node name, we have to do line wrapping ourselves. So we split lines using the current terminal width. `roslaunch` does not do that, they don't interfere with the printing at all. In their case, the terminal (or GNU screen) does the line wrapping, so when you change the terminal width, your terminal can re-wrap all the lines.

Another cornerstone of rosmon's design is that it doesn't take control of your terminal. Many ncurses applications take full control and enter 'alternate' mode, where you cannot use the scrolling facilities of your terminal anymore. This is hugely annoying especially for an utility like rosmon, which mainly produces log output. But since we avoid alternate mode, we cannot be sure what the terminal content actually looks like. This means that we cannot go back and re-write log lines. Once they are written, they stay the way they are, even if the terminal changes size...

I can see some ways to solve this:
 * Make rosmon use `alternate` mode and fully implement scrolling and searching the log history. In that case, we could simply 
    re-render the screen with new dimension when the terminal width changes. I think many users want this feature and it would 
    make rosmon more powerful. But it's a huge task to to implement and disables terminal scrolling as mentioned above.
 * Make GNU screen somehow stay with the last terminal width instead of switching to 80.
   Or use `tmux` instead, which seems to do that by default.
 * Make rosmon detect this situation and implement some special handling.
 * ...?

zlacelle, 2020-10-28:
Understood. I can try and dive into the default non-adjusted terminal size settings for screen or xterm or whatever the underlying terminal is. It wasn't very obvious initially how to change this.

Closing since this seems tough to change.

closed 2020-10-28 by zlacelle