xqms/rosmon #135: Add "view all nodes" to rqt_rosmon GUI

zlacelle, 2020-10-28:
If we've got a bunch of rosmon launches with different launch files, it would be good to be able to view all running nodes and their corresponding CPU/Memory usage and state in one place. Can the rqt_rosmon gui be enhanced with a "view all" tab?

xqms, 2020-10-28:
It certainly can be done. Here's what I would do:

* add a special [all] key to `ROSMonModel`
* add a slot to `MonGUI` that listens to `ROSMonModel` updates and creates subscribers for each rosmon topic
* extend `NodeModel` to save a "source topic" for each entry, and modify the `NodeModel::updateState()` method:
  * add a parameter "sourceTopic"
  * add logic that adds/removes nodes belonging to that sourceTopic.

I won't have time for this in the near future. But as always, PRs are welcome!

zlacelle, 2020-10-28:
Makes sense. I can probably tackle this slowly over the next few months. Thanks!