xqms/rosmon #136: Cannot build from source: did you forget to ‘#include <string>’?

andrewjong, 2020-11-09:
Cloned `rosmon` and `rosfmt` under my `src/`. Added `set(CMAKE_CXX_STANDARD 11)`.

I get this during `catkin build`.

```
Errors     << rosmon_core:make /home/andrew/Development/subt/communication_manager/logs/rosmon_core/build.make.000.log                                                                                            
In file included from /home/andrew/Development/subt/communication_manager/src/rosmon/rosmon_core/src/launch/bytes_parser.cpp:4:
/home/andrew/Development/subt/communication_manager/src/rosmon/rosmon_core/src/launch/bytes_parser.h:19:12: error: ‘uint64_t’ was not declared in this scope
   19 | std::tuple<uint64_t, bool> parseMemory(const std::string& memory);
      |            ^~~~~~~~
/home/andrew/Development/subt/communication_manager/src/rosmon/rosmon_core/src/launch/bytes_parser.h:19:26: error: template argument 1 is invalid
   19 | std::tuple<uint64_t, bool> parseMemory(const std::string& memory);
      |                          ^
/home/andrew/Development/subt/communication_manager/src/rosmon/rosmon_core/src/launch/bytes_parser.h:19:51: error: ‘string’ in namespace ‘std’ does not name a type
   19 | std::tuple<uint64_t, bool> parseMemory(const std::string& memory);
      |                                                   ^~~~~~
/home/andrew/Development/subt/communication_manager/src/rosmon/rosmon_core/src/launch/bytes_parser.h:8:1: note: ‘std::string’ is defined in header ‘<string>’; did you forget to ‘#include <string>’?
    7 | #include <tuple>
  +++ |+#include <string>
    8 | 
make[2]: *** [CMakeFiles/rosmon_launch_config.dir/build.make:147: CMakeFiles/rosmon_launch_config.dir/src/launch/bytes_parser.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1045: CMakeFiles/rosmon_launch_config.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
```
Simply adding `#include <string>` in `bytes_parser.h` made these errors disappear. How come this issue appeared?

xqms, 2020-11-10:
Hey, thanks for the report. These kind of problems can happen due to differences in the standard library or compiler flags - apparently on Ubuntu 16.04/18.04/20.04 the `<string>` header is included by another already included header, which is why everything compiles just fine for me.

Can you tell me your Linux distribution and compiler version (`gcc --version`)?

Adding `set(CMAKE_CXX_STANDARD 11)` should also not be necessary.

xqms, 2020-11-10:
I'm still curious about your system details, but I just pushed a fix in #137, so I'll close this issue.

closed 2020-11-10 by xqms

andrewjong, 2020-11-10:
Thanks a lot! I'm on Arch Linux.

PR #137 (2020-11-10): core: launch: bytes_parser: add missing include
Fixes #136. Thanks to @andrewjong for reporting. Somehow on the CI machines and the usual systems I compile on `<tuple>` seems to include `<string>`. Weird...