xqms/rosmon #142: terminal input

akashjinandra, 2020-12-10:
Hello, is it possible to accept cin input via terminal when running multiple nodes?

xqms, 2020-12-10:
Hey, rosmon currently takes control of the keyboard, no application can get data via stdin (it doesn't matter if it is multiple nodes or a single one). I *could* imagine a feature where you select a node and request that you will be connected directly to the node - with some sort of escape key combination to get back to rosmon. But this is not something I need, so if you need it, please write a PR :)
But in my opinion, interactive nodes should be started outside of rosmon.

akashjinandra, 2020-12-14:
Thanks for the input. I will look into this!

closed 2020-12-14 by akashjinandra