xqms/rosmon #143: Feature Request: Support for ROS2

d-walsh, 2021-01-15:
It would be great to if rosmon could work with ROS2.

xqms, 2021-01-15:
Yes that would be great indeed. Since I'm not using ROS2 (yet), I don't have any motivation to work on it, though. Here are some thoughts that might be relevant to someone looking into this:

* The rosmon project started in a pretty special situation: My team for the DARPA Robotics Challenge needed a way to supervise large amounts of nodes, restart nodes dynamically, *and be sure the monitoring system would not crash*. Since no one of us was very proficient in Python (and I still think it's a bad idea to write mission-critical stuff in dynamic languages), this led to a complete rewrite of the roslaunch logic in C++. While this had unexpected benefits (I learned a lot about the hidden intricacies and bugs in the roslaunch XML parser), I think that bolting a python-based ncurses UI onto roslaunch would also have been a solution, and this may very well be a better/easier solution for ROS2.
* It seems that the launch system in ROS2 is even more python-based than in ROS1 - capable of generating node constellations dynamically. I have no idea how well an alternative launcher like rosmon can be adapted to work with it. *If* the python part can be used in a way that spits out command lines for processes and rosmon can then take over the monitoring part, it could work.
* Apart from the launch file parser, which would need to be adapted to whatever ROS2 uses, the ROS interface (topics/services) of rosmon would also need to be ported.
* Because of the tight integration with roscpp, most parts of rosmon would need to be touched for a port of ROS2. So it might be better to fork (or completely rewrite) rosmon for ROS2.

I'm going to leave this issue open to collect information and interested users as well as contributors. As stated above, I won't invest my time into ROS2 (yet).

buckleytoby, 2023-04-28:
@xqms any updates on this thread?

xqms, 2023-05-01:
Nothing has changed. I'm not developing on ROS2 (yet) so I have no motivation myself to do something. Contributors are welcome, though!

gibsonhu123, 2024-03-12:
I really need this :( , I refused to migrate to ros2 until rosmon works. Maybe somebody just use ai to do the migration O.o

Dhagash4, 2024-09-23:
@xqms Thanks for an amazing open source project in ROS1. As I am predominantly using ROS2 would like to have a similar system for monitoring nodes launched in ROS2 as well, would be great contribution.

What I think would be amazing is to monitor all the nodes visible via your DDS and show whether they are active/inactive with nice colors as seen in rosmon and some statistics like CPU/GPU usage. I am willing to contribute to a new branch starting with something simple as this, and then we can see where it goes! However, I will need some guidance and pointers on whether this is feasible and in general discussion about this feature! Let me know what are your thoughts on it, Have a nice week!

ndahn, 2025-07-31:
@xqms I hope you're fine with me posting this here. If it's a problem I'll be happy to remove this post again.

For anyone interested in a solution for ROS2, I recently released [better_launch](https://github.com/dfki-ric/better_launch) which comes with an optional terminal UI.

xqms, 2025-07-31:
@ndahn No problem! I put a link to your project in the README :)