xqms/rosmon #147: logwarn, logerr behavior

leviathan747, 2021-06-28:
Hi, I'm having a bit of an issue with logging. I'm running a node using Python 3 (need a `python3` only module) and using `asyncio`. I was getting all my "INFO" logs in the monitor, but none of the "WARN" or "ERROR" level logs.

A couple of things I noticed/confirmed during debugging:
- All of the logs were getting published to `/rosout` (confirmed with `rostopic echo`)
- I am still able to get all log types in my Python 2 nodes, however they are all colored with white text (not red or yellow to indicate level; see screenshots to see difference between `roslaunch` and `rosmon`)

[image]
[image]

I'm sure there's something different about how rosmon captures the logs, but I wasn't able to find documentation anywhere. Can I get some help? I could make all my logs info, but I'd prefer not to...

xqms, 2021-07-02:
Thanks for reporting, that is definitely a bug. I was able to reproduce this on my machine (with Python 3). I'll look deeper into it, I suspect the python nodes are not properly flushing their stderr for some reason, because the messages appear as soon as the node exits.

xqms, 2021-07-06:
Should be fixed now in master. Could you please test if both issues are gone now?

leviathan747, 2021-07-09:
It works (ROS Melodic on Ubuntu 18.04). Will this be in a melodic release soon?

xqms, 2021-08-02:
Thanks for the feedback. I'll create a new release today.

closed 2021-08-02 by xqms