xqms/rosmon #150: YAML merge key parsing not correct under lists/array environment

QiayuanLiao, 2021-07-23:
Hello, I found that the YAML merge key is working well under the struct environment but not correct under the lists/array environment.
###  YAML code:
```
  defaults: &defaults
    A: 1
    B: 2
  struct:
    <<: *defaults
    A: 23
    C: 99
  array:
    - <<: *defaults
      A: 23
      C: 99
```
### roslaunch outupt:
```
array:
- A: 23
  B: 2
  C: 99
defaults:
  A: 1
  B: 2
struct:
  A: 23
  B: 2
  C: 99

```
### mon launch output:
```
array:
- '<<':
    A: 1
    B: 2
  A: 23
  C: 99
defaults:
  A: 1
  B: 2
struct:
  A: 23
  B: 2
  C: 99
```
### Difference:
Note the  causing by the lists/array environment
```
- '<<':
    A: 1
    B: 2
```

xqms, 2021-07-27:
Hi @QiayuanLiao, thanks for the report! Any deviation from roslaunch is definitely a bug.

Merge keys are not official YAML syntax and are not supported by the YAML parser we are using (yaml-cpp). So @Cartoonman implemented merge key handling inside rosmon... See #117 for details.

I guess a small fix is needed, but I don't have time to look at it at the moment. If you can come up with a PR I'd be happy to look at it, though.

xqms, 2021-08-02:
PR #153 should fix your issue. Could you please test it?

PR #153 (2021-08-02): core/launch: support YAML merge keys inside arrays
This should fix #150, where problems with the non-standard YAML merge keys `<<` inside arrays were noticed:

```yaml
defaults: &defaults
  A: 1
  B: 2
array:
  - <<: *defaults
    A: 23
    C: 99
```

Previously, merge keys where only handled on a per-parameter level. Since arrays are imported onto the ROS parameter server as a single XmlRpcValue, merge keys inside were not handled at all. We now do that inside `yamlToXmlRpc()`, which fixes this situation.

It also adds a rudimentary unit test.
closed 2021-08-02 by xqms