xqms/rosmon #159: Could not find a configuration file for package "Qt5Widgets" that exactly matches requested version "5.10.1"

Emmanuel-Messulam, 2021-09-17:
I cannot get this module running with an existing project, I am getting this error when building with `catkin_make`:

```
CMake Error at rosmon/rqt_rosmon/CMakeLists.txt:33 (find_package):
  Could not find a configuration file for package "Qt5Widgets" that exactly
  matches requested version "5.10.1".

  The following configuration files were considered but not accepted:

    /usr/lib/x86_64-linux-gnu/cmake/Qt5Widgets/Qt5WidgetsConfig.cmake, version: 5.12.8
    /lib/x86_64-linux-gnu/cmake/Qt5Widgets/Qt5WidgetsConfig.cmake, version: 5.12.8

-- Configuring incomplete, errors occurred!
```

xqms, 2021-09-17:
Hey, I'm quite sure I know what's going on: You probably compiled your workspace, then did a system upgrade, and now you're trying to compile again. In the meantime, your Qt version has changed, which is confusing the build system.

Can you try removing your `build` folder and compiling again?

If that doesn't fix your problem, can you share your OS version (Ubuntu X.Y) and your ROS version?

Emmanuel-Messulam, 2021-09-20:
> Can you try removing your `build` folder and compiling again?

I tried that, same error.

> If that doesn't fix your problem, can you share your OS version (Ubuntu X.Y) and your ROS version?

Ubuntu 20.04, ROS noetic.

Emmanuel-Messulam, 2021-09-20:
Nevermind, the version hadn't updated correctly.

closed 2021-09-20 by Emmanuel-Messulam