xqms/rosmon #165: Error when running rosmon inside launch file

igricart, 2022-06-29:
I have this error when running the system after building from source
```
Traceback (most recent call last):
  File "/home/imr/Documents/ws/tmp/test/ps_ws/devel/lib/rosmon/rosmon", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/imr/Documents/ws/tmp/test/ps_ws/src/third_party/rosmon/rosmon/src/rosmon", line 2
    exec rosrun rosmon_core rosmon "$@"
```

when my launch file contains
```
<node pkg="rosmon" type="rosmon" output="screen" name="mon_bag" args="--flush-log --no-start --disable-diagnostics --disable-ui --name=mon_bag --log=/tmp/rosmon/mon_bag.log $(find belt_loader_bringup)/launch/recorder.launch/>
```

xqms, 2022-07-01:
Thanks for the report! That's definitely a bug with the installation of the `rosmon` wrapper script. I fixed it in #166, which will be merged as soon as the CI is finished.

As a workaround, you can also use `pkg="rosmon_core" type="rosmon"` which avoids the wrapper script and runs the rosmon node directly.

xqms, 2022-07-01:
#166 is merged, so I'll close this. Feel free to reopen if the issue is still present.

closed 2022-07-01 by xqms