xqms/rosmon #167: Start Stop All nodes

renan028, 2022-07-01:
I have created this fork: https://github.com/xqms/rosmon/compare/master...renan028:rosmon:start_stop_all
that adds the feature to start/stop all nodes if one calls the action with an empty string. 
Let me know if you find it useful. Or maybe change to a specific keyword (dangerous conflict).

Case:
You have several nodes with high cpu consumption, and those don't need to be alive most part of the time. Add those to a launch file and run it with rosmon. Then you can call the service to stop/start all those nodes from a BehaviorTree (or other executive) when you don't/do need them.

xqms, 2023-01-25:
Hey, I'm sorry for the *very* long silence, somehow I missed this issue. That sounds interesting, although I would like a separate `start_all`/`stop_all` service better rather than modifying the existing one. It may be unexpected that you stop all nodes if you simply call the service with an empty/default argument.

Would you like to open a PR for this?

renan028, 2023-01-25:
Hey, that makes sense to me. 
I will make a PR with those changes.

renan028, 2023-01-27:
Hey, what do you think about adding a new `boolean` field `all_nodes` in the existing `srv`:
```
uint8 START = 1
uint8 STOP = 2
uint8 RESTART = 3

string node     # ROS node name
string ns       # ROS node namespace
uint8 action
bool all_nodes  # if True call the action to all nodes in given namespace
---

```
Otherwise, we will create a new `srv` that is basically identical to the old one `StartStopAll`

xqms, 2023-01-27:
Just an idea I had: What if we allow regexes in the `node` field?

Specifying `node=".*"` would allow starting/stopping all nodes. `/abcde/.*` would only affect nodes in the `abcde` namespace. And so on...

The logic would look like this:

1. If `ns` is not empty, prepend it to the regex
2. For each node, check if its fully-qualified name (including the namespace) matches the regex.

Step 1 keeps backward compatibility with previous versions.

renan028, 2023-01-30:
yep. I liked the idea. Regex is way more powerful.
I will give it a try this week

AngeLoGa, 2023-12-15:
Any updates on this? I thought this was in use but after checking the srv message I saw no changes and no mention about either RE or bool for all. I thought the field "ns" worked similar to:

> `/abcde/.*` would only affect nodes in the `abcde` namespace

But I understood it uncorrectly. I feel bad for not being able to contribute if needed but my knowledge in C++ is limited yet.

Thanks from advance!

xqms, 2023-12-15:
Hey @AngeLoGa, so far nobody has implemented my proposal above https://github.com/xqms/rosmon/issues/167#issuecomment-1406859715 yet. If you want to try, go ahead :)

This is the spot where it would need to be done: https://github.com/xqms/rosmon/blob/12a4934bb6a6057529eda06ecc190c16cacb52d6/rosmon_core/src/ros_interface.cpp#L75

renan028, 2023-12-15:
Sorry, I totally forgot to PR this: https://github.com/xqms/rosmon/pull/183

xqms, 2024-02-19:
Finally got around to merging #183 - sorry for the long delay. I will close this issue accordingly.

closed 2024-02-19 by xqms