xqms/rosmon #170: output="log" is ignored with --disable-ui

Rayman, 2022-11-16:
I'm using `output="log"` in my launch files because I want the stdout to only go to the log file (not printed to the screen). But when I use the option `--disable-ui` it seems this attribute is ignored and the output is still printed to the screen.

#### How to reproduce
`example.launch`
```
<launch>
    <node name="rostopic" pkg="rostopic" type="rostopic" args="list" output="log"/>
</launch>
```
Launching with UI works
```
rosrun rosmon rosmon example.launch 
Loaded launch file in 0.007098s
ROS_MASTER_URI: 'http://localhost:11311'
roscore is already running.
Running as '/rosmon_1668607956691471995'

           /rostopic: /rostopic exited with status 0
            [rosmon]: Shutting down...
```
And launching with `--disable-ui` you can see the stdout printed to the screen
```
rosrun rosmon rosmon example.launch --disable-ui
Loaded launch file in 0.007307s
ROS_MASTER_URI: 'http://localhost:11311'
roscore is already running.
Running as '/rosmon_1668607924412738905'

/rostopic: /diagnostics
/rostopic: /rosmon_1668607924412738905/state
/rostopic: /rosout
/rostopic: /rosout_agg
/rostopic: /rostopic exited with status 0
```

xqms, 2022-12-09:
Should be fixed in master, thanks! Let me know if the issue still persists.

closed 2022-12-09 by xqms

Rayman, 2022-12-13:
Thank you! I've tested it and I can confirm it now works.

PR #171 (2022-12-09): core: main: hide stdout log events if output=screen and UI is disabled (#170)
Should fix #170.