xqms/rosmon #178: Std out issue since latest noetic update

alaurenzi, 2023-09-29:
Hi Max,
I am experiencing an issue regarding std out prints, I believe since the last noetic update. It looks somehow related to ASCII formatting codes. Apparently, bold formatting is not showing up, as well as anything between the end of bold formatting and newline.

Check the last three lines of these pictures:

**Rosmon compiled from source, checked out at 4f9b10e (works ok)**
[image]

**Rosmon compiled from source, latest master (broken output)**
[image]

I'll try to provide a minimal example reproducing this issue, in the meantime maybe this already rings a bell to you!

Thanks in advance,
Arturo

xqms, 2023-09-29:
Hey Arturo, interesting... The color coding stuff is pretty complicated, since rosmon needs to parse all codes and remember them. There might be a line from another node in between, so rosmon parses and restores all color coding for each node. Probably you guys are using an escape code that the current version doesn't handle correctly...

Are you using a library for the logging? Or is that custom code? It would be great if we could find out which escape codes you are outputting. Maybe launch the `xbot2` node with `rosrun ... &> log.txt` and paste the log file here?

xqms, 2023-09-29:
I found at least one bug in the parser and fixed it in master. Can you compile current master and test if it fixes your case as well?

alaurenzi, 2023-09-29:
Thank you Max, it seems it works now!