xqms/rosmon #180: Core Dumped for too many open files

tugbakara, 2023-10-02:
When  I run rosmon sometimes it gaves error and closed itself randomly:
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not create stderr pipe: Too many open files
Aborted (core dumped)
```

xqms, 2023-10-02:
Hm, that can point to a bug in rosmon (file descriptor leak). Some questions for you:

 * How many nodes are in your launch file?
 * Can you post the output of `ulimit -n`? If that limit is too low, than rosmon cannot work properly.

hsynklceng, 2023-10-09:
There are 16 nodes in my launch file. 
ulimit -n is 1024.

xqms, 2023-10-12:
I think we need more information. Can you share your launch file? Or try to reduce it to a minimal example that reproduces the problem?