xqms/rosmon #182: Add options for console logging Node name right justification

mitch-galea, 2023-12-14:
Hello,

Some of the node names in our launch file are very long, with the right justification of ros node names and paired with journalctl logs means that the output can look pretty ugly, especially for the nodes with short names. 
[image]

We have noticed the issue when updating from 2.3.2 -> 2.5.1. Below is what we think is the cause:
[image]

Could we add an option not to right justify and align all the logs or set a max_length value manually rather than computing it based on the longest node

xqms, 2023-12-14:
Just to make sure: You care about the justification in *journald*, right? The code snippet in your screenshot is about stdout.

Maybe we should switch off the alignment in the syslog/journald output altogether? Not sure if anyone cares about that there.

xqms, 2023-12-14:
Ah, now I understand: You are running rosmon from systemd and so the stdout goes to journald. Hm, yes, I guess we would need an option for that.

An alternative for you would be to switch off stdout forwarding in systemd service:
```
[Service]
...
StandardOutput=null
```

(reference: https://www.freedesktop.org/software/systemd/man/latest/systemd.exec.html#StandardOutput=)

Instead, you would enable rosmon's own logging system by specifying the `--log=syslog` parameter. That system does not justify/align anything :)

closed 2024-01-18 by mitch-galea